f1xpl/openauto #1: Nice project

Vitalic66, 2018-02-14:
Hi,

just found your project from a comment you left in an older thread i started. 

The video looks pretty cool. Seems to run very smooth.

Any install instructions? Which image do you use on Pi3? Which dependencies are needed?

What kind of touchscreen do you use in the video?

Regards

f1xpl, 2018-02-14:
Hello,
here you can find build instructions for the OpenAuto https://github.com/f1xpl/openauto/wiki/Build-instructions .

You need to install dependencies (all dependencies are listed in OpenAuto and aasdk build instructions), build aasdk and then build OpenAuto.

The easiest way is to use Raspbian Stretch. Raspbian Stretch has preinstalled GCC 6.3.0 with support of C++14 standard.

For the video I have used official Raspberry PI 7 inches touchscreen.

Vitalic66, 2018-02-14:
Thx for information. Will give it a try. Already got all things needed here :) .

f1xpl, 2018-02-14:
Good luck. Let me know whether everything works. Please remember to follow also instruction regarding USB permissions. Here is the link https://github.com/f1xpl/openauto/wiki/udev-rules-(USB-permissions)

Vitalic66, 2018-02-14:
Yeah this was one of the major problems on another project i tested.

Maybe there will be one problem for me. I use the pibus hifiberry dac as sound output. Guess i have to configure PA to use it as sink and usb sound card as source for mic input.

f1xpl, 2018-02-14:
It uses Qt library for the audio output and audio input. As far as I know Qt internally uses Pulse Audio. In my case I just connected USB sound card and everything worked like a harm without any additional configuration changes.

If there is no option to connect mic to the hifiberry (I am not familiar with it honestly) then I think you should play a bit with the configuration of pulse audio.

Vitalic66, 2018-02-14:
Have you ever thought about Wifi connection instead of USB?

In the other project it worked for Android Auto.

Vitalic66, 2018-02-14:
You may create a gitter room?

f1xpl, 2018-02-14:
Here is the link https://gitter.im/openauto_androidauto

I will add it to the Readme later.

f1xpl, 2018-02-16:
All topics have been clarified.

closed 2018-02-16 by f1xpl