f1xpl/openauto #2: OBD-ii Support

dj21d, 2018-02-14:
Hi there,
I'm looking forward to implementing your project into my vehicle. Does your project currently or do you plan to support OBD-ii diagnostics on the app? I have a bluetooth OBD-ii receiver that I would love to integrate with this.

f1xpl, 2018-02-16:
I hope it will be opportunity to look on this at 1.2 version.

akileos, 2018-02-18:
But the OBD-II should be on android side, no Pi-Side. What would be the benefit ?

rapzak, 2018-02-19:
I have a keypad in my car i want to use as input, it serves the signal on CAN bus (multimedie can bus)

Also i need to switch amplifier in my radio to select sound it also has input for navigation etc there maybe could be used smart.

umer936, 2018-02-22:
Wanted to subscribe to this. I've been using an RPi in my glove compartment for logging OBD-II data for a few years now (as I posted about here: https://www.reddit.com/r/raspberry_pi/comments/3782a3/obdii_pi/ ). 

For this though, it may be better to have the OBD-II connect to the Android side and pass through to the screen. Though, that sounds a lot more difficult. Plus, having it on the RPi would split up the jobs between the devices, hopefully making things faster than everything running on the Android.

pgwipeout, 2018-02-22:
Considering the Pi has the headers to accomplish this, and most head units these days have a canbus line fed to them, it seems reasonable to pull off.
The issue I see is this is not a native supported feature of Android auto, so it would have to be implemented as a separate service.
So I guess this should be split up into two features, this one being the hardware support for canbus, and another for plugin support for additional software running alongside the Android auto service.

szeater, 2018-02-22:
I would suggest, given the above, that it be its own separate project then. The only way to interface with AA directly is via an Android application, which would mean having the OBD connection to the phone. The primary purpose of this project is to run AndroidAuto on the RPi. No more, no less.

rapzak, 2018-02-22:
Hi, I don't agree with this, there must be a way to interact with the car
for build in joystik for interaction, which is present on the multimedia
canbus - also if there needs some sort of communication with the radio for
opening AUX channel etc.

But for sure, if we talk car diagnose etc. then i agree it will be atleast
second priority.

Also i could think if there could be made some short of SW-interface where
this code could be sepperate from this project, and maybe could be coded in
different language as python etc.

This is my oppinion, with small knowledge to the AA, but it is logical the
control of the unit goes directly.

/Kasper

pgwipeout, 2018-02-22:
Considering the fact that some vehicles implement canbus controls of the head unit,.
I think the best course of action is to create an extension API into this project, and let someone else write the extensions in a separate project.

rapzak, 2018-02-23:
Hi, and the can interface would properly be different from unit to unit... - but i would agree :)

Vitalic66, 2018-02-23:
BMW E models like mine for example use IBUS... I think too that this project should stay standalone. I just need this to get AA working in Pi. Everything else is done with some other apps i already use for the IBUS. Maybe there is something similar for your car like for mine www.pibus.info . Right no i'm able to listen to the IBUS with this extra hardware. I can read and write to the IBUS with shell commands.

Vitalic66, 2018-02-23:
No, shell commands and dbus are enough. I already used that to control musicplayer via BT via keypress on stearingwheel. For example: keypress on stearingwheel->IBUS message->cached by pibus (translated to configured keypress)->shell with dbus message to BT->command via BT AVRCP to phone

aledthom, 2018-02-25:
Can't you use can bus interface to generate key presses? Open auto already supports keyboard shortcuts so it doesn't need to integrate this within the app.

closed 2018-03-01 by f1xpl