f1xpl/openauto #3: Support of TCP transport

f1xpl, 2018-02-15:
Support TCP transport in order to use AndroidAuto wireless.

f1xpl, 2018-03-18:
> 03-18 00:48:58.882 17691 17691 W CAR.WIFI.INFO: Unsupported GmsCore version for wireless projection.

It looks like AA wireless is not yet officialy supported. Merging latest changes of feature/wireless to the development branch and postponing development of this feature until official support will be available.

f1xpl, 2018-03-25:
Implementation of wireless access using head unit server has been merged to the development branch. Still waiting for the Google Play Services capabilities to use bluetooth connection as a wifi credentials exchange method.

parkerlreed, 2018-03-26:
[image]

Love it! Occasional audio glitch where it goes higher pitched but it goes away and becomes stable (for as long as I was testing)

martenumberto, 2018-04-11:
So is wireless connection supported?