f1xpl/openauto #5: Recommanded hardware

akileos, 2018-02-18:
Hello,

 Could your add a wiki page or add the "Known working" hardware ? 
Supported Pi Version, touchscreen, sound card (you need one for mic,right?)

Thanks.

f1xpl, 2018-02-18:
In case of supported hardware, everythings that is supported by RPI 3 is supported by OpenAuto as well.

Right now the only tested version is Raspberry Pi 3 B+.

However, it is good idea to create such wiki topic based on the community feedback.

shawngmc, 2018-03-01:
This is especially important for touchscreens.

Many cheap touchscreens on Amazon, for example, work with the RPi3, but only if you use their distro which has a binary blob driver.

Just to be clear, which touchscreen have you used already?

eltonkola, 2018-03-05:
would this run on Raspberry Pi 2 Model B Rev 1.1 ?

gitmayer, 2018-03-26:
Did you have to do something special on the Pi 2? I don't know if I am having this issue because of the board or how I have it set up.  But I can't get it to run on official Pi touchscreen, getting this error:

[2018-03-26 18:51:56.771740] [0x726dc250] [warning] [AaApp] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
QXcbConnection: Could not connect to display

gitmayer, 2018-03-27:
Nevermind the above comment. RTFM, right?

closed 2018-04-05 by f1xpl