f1xpl/openauto #7: No sound

inialsien, 2018-02-19:
I can hear listen music by VLC normally. But no sound when run with openauto.

Here is my console log. 
```text
[2018-02-19 11:38:05.509097] [0x693dc310] [info]    [AaApp] [AudioService] stop indication, channel: MEDIA_AUDIO, session: 0
[2018-02-19 11:38:17.974569] [0x67bdc310] [error]   [AaApp] [VideoService] channel error: AaSdk error code: 10, native code: 1
[2018-02-19 11:38:17.974735] [0x683dc310] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: MEDIA_AUDIO
[2018-02-19 11:38:17.974824] [0x683dc310] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SYSTEM_AUDIO
[2018-02-19 11:38:17.974895] [0x683dc310] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SPEECH_AUDIO
[2018-02-19 11:38:17.974975] [0x683dc310] [error]   [AaApp] [AudioInputService] channel error: AaSdk error code: 10, native code: 1
[2018-02-19 11:38:17.975079] [0x683dc310] [error]   [AaApp] [BluetoothService] channel error: AaSdk error code: 10, native code: 1
[2018-02-19 11:38:17.975168] [0x68bdc310] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 10, native code: 1
[2018-02-19 11:38:17.975231] [0x67bdc310] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 10, native code: 1
[2018-02-19 11:38:17.975365] [0x693dc310] [error]   [AaApp] [AndroidAutoEntity] channel error: AaSdk error code: 10, native code: 1
```

f1xpl, 2018-02-19:
Please check whether pulseaudio is installed, up and running.

inialsien, 2018-02-19:
Problem solved.  Problem caused by `module-echo-cancel`. 

Thank you very much.

```text
Feb 19 09:00:19 pi3 pulseaudio[1237]: [pulseaudio] module-echo-cancel.c: Can't cancel echo between a sink and its monitor
Feb 19 09:00:19 pi3 pulseaudio[1237]: [pulseaudio] module.c: Failed to load module "module-echo-cancel" (argument: "aec_method=webrtc aec_args="mobile=1 digital_gain_control=1 analog_gain_control=0 comfort_noise=0 noise_suppression=1 routing_mode=loud-speakerphone" source_name=echoCancel_source sink_name=echoCancel_sink"): initialization failed.
Feb 19 09:00:19 pi3 pulseaudio[1237]: [pulseaudio] main.c: Module load failed.
Feb 19 09:00:19 pi3 pulseaudio[1237]: [pulseaudio] main.c: Failed to initialize daemon.
Feb 19 09:00:19 pi3 pulseaudio[1234]: [pulseaudio] main.c: Daemon startup failed.
```

closed 2018-02-19 by inialsien