f1xpl/openauto #8: openauto.ini

dev-danny, 2018-02-19:
Hi so finally got it installed and it loaded up. it still says waiting for device even when I have plugged my device in with and without Android Auto running. But as I look at the logs before it loads it says 

`[2018-02-19 19:52:07.123879] [0x7280d000] [warning] [AaApp] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file
libEGL warning: DRI2: failed to authenticate

** (autoapp:1058): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
[2018-02-19 19:52:08.851680] [0x693ff310] [info]    [AaApp] [USBApp] Waiting for device...
`
Any advice on this issue?

Thanks

f1xpl, 2018-02-19:
Did you set udev rules?

dev-danny, 2018-02-19:
hi, I set up udev rules as explained in your wiki and I changed the GPU memory to 256MB and the GL drives as you suggested.

this is the logs before it boots openauto
`pi@raspberrypi:/opt/vc/src/hello_pi/libs/ilclient/openauto/bin $ ./autoapp
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information

** (autoapp:942): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
[2018-02-19 21:16:15.095466] [0x6b3dc310] [info]    [AaApp] [USBApp] Waiting for device...
QXcbConnection: XCB error: 148 (Unknown), sequence: 174, resource id: 0, major code: 140 (Unknown), minor code: 20
`
it boots but like before just says the waiting for device.

my device is a Samsung S8 with Oreo 

thanks

f1xpl, 2018-02-20:
Did you run autoapp via SSH or directly on PI?

dev-danny, 2018-02-20:
Via VNC/SSH. I see the waiting for device screen and can change settings and cursor etc.

I have noticed on my phone that in the AA  settings, openauto shows up as a connected car

f1xpl, 2018-02-20:
You must execute autoapp directly on PI. Not via SSH or VNC.

dev-danny, 2018-02-22:
hi, I have tried all you have suggested. changed GL drivers on raspi-config, executed it locally. It loads up and says waiting or device and when I plug my phone in the android app loads and I can set it up but the screen on the raspberry pi doesn't change. my phone shows its connected to a car called openauto and my phone screen is black and says Android Auto on it but as I say the raspberry pi screen doesn't change. any ideas?

I downgraded my OS from Oreo to Nougat  on my phone to see if that helps but it doesn't

f1xpl, 2018-02-22:
What is your video output? HDMI, touchscreen or RCA?

dev-danny, 2018-02-22:
Hi I dunno what I did but I turned my Pi on again today loaded openauto and then connected my phone and it worked :). I have no idea whats changed but I'm gunna leave it alone now. I was using hdmi but now it works on touchscreen aswell

thank you again

closed 2018-02-22 by dev-danny