f1xpl/openauto #9: Audio Routing Logic

mbt28, 2018-02-20:
Hi,

Can you please explain how the audio background working?

How the application receive notification sounds, speech/phonecalls, music? 
Accordingly I will choose to make output trough hdmi or I2S.

Thanks

f1xpl, 2018-02-22:
Audio playback is handled by the Qt multimedia library. According to the Qt documentation it uses pulseaudio as a backend on Linux.

mbt28, 2018-02-23:
Allright, but can all the audio be routed through usb or is bluetooth obligotory to have "notifications, and phone calls"? And how are the audio channels of the streams? For example is the android streams stereo audio, and one separated channel for notifications+phone calls?

Is the audio stream of music pcm format or is it a media format?

Thanks

f1xpl, 2018-02-23:
Media, Speech and Notifications audio channels are routed through the USB, phone calls are router through the Bluetooth (HFP profile).

Media is 48 KHz stereo.
Notifications and Speech are 16KHz mono.

closed 2018-02-25 by f1xpl

mbt28, 2018-03-02:
Hi,

Why AA doesnt route the phone audio trough usb audio delivery profile? It is because the current slave usb audio delivery protocol doesnt support more than 2 channels?

Thanks