f1xpl/openauto #11: Enable Browsing

Vitalic66, 2018-02-22:
So you think it is possible to enable browsing media from client side?

Or is it a server side problem (AA app on smartphone)?

AA is nice but this browsing problem sucks.

Is browsing disabled when sensors read wheel movement?

szeater, 2018-02-22:
I'm *pretty* sure this is an AA app issue. Even when running standalone on the phone, once it starts picking up acceleration/rapid movement, it stops you from interacting with apps properly.

f1xpl, 2018-02-22:
Browsing logic is handled in AA app. It looks like head unit has nothing to do with it.

mbt28, 2018-02-23:
Hi,

Maybe it can be tricked by head unit with sending manuplated data but I beleive it is a good future to provide safer cruising.

closed 2018-02-25 by f1xpl