f1xpl/openauto #16: Implement configuration of OMX layers

f1xpl, 2018-02-24:
According to [this pull request](https://github.com/f1xpl/openauto/pull/15), configuration for the OMX display layer is needed.

TODO:
-Extend [IConfiguration](https://github.com/f1xpl/openauto/blob/development/include/f1x/openauto/autoapp/Configuration/IConfiguration.hpp) interface for setters and getters of the layer value
-Add field at Settings GUI (video tab) to set layer value
-Use the configured value of layer in [OmxVideoOutput](https://github.com/f1xpl/openauto/blob/development/src/autoapp/Projection/OMXVideoOutput.cpp) class

f1xpl, 2018-03-03:
Changes done on develop branch - [commit](https://github.com/f1xpl/openauto/commit/5984108d2009ec69f3eedc9efff82bf8f42ca2c5). Waiting for merging to master.

parkerlreed, 2018-03-03:
Something I've noticed. Develop branch seems to be default. If I clone the repo with the base URL, develop seems to be the "only" branch.

```
[parker@e55 build]$ git clone https://github.com/f1xpl/openauto
Cloning into 'openauto'...
remote: Counting objects: 341, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 341 (delta 1), reused 10 (delta 1), pack-reused 321
Receiving objects: 100% (341/341), 97.07 KiB | 2.06 MiB/s, done.
Resolving deltas: 100% (189/189), done.
[parker@e55 build]$ cd openauto
[parker@e55 openauto]$ git branch
* development
```

Figured I'd give a heads up in case this wasn't intentional.

f1xpl, 2018-03-03:
According to the correct branch flow - development branch should be default. Here is good [tutorial](http://nvie.com/posts/a-successful-git-branching-model/) about it.

f1xpl, 2018-03-03:
Merged to master.

closed 2018-03-03 by f1xpl