f1xpl/openauto #18: Cracks in audio stream

Vitalic66, 2018-02-25:
From what we discussed on gitter, I tried different Settings in PA config, all without success. So today as last option i defined the default sample rate to the lowest from google voice as you said it is on 16000Hz.

Only this way cracks are gone. But unfortuneatly the audio stream sound is crap now. Is there a possibility to optimize the mix of audiostreams in Qt? I guess when playback of audio with default 48000Hz and voice with 16000Hz the mixed stream produces the cracks and the stuttering.

Vitalic66, 2018-02-25:
Or is it possible to output 2 different streams so i can mix them after Qt in pulseaudio. Right now PA receives a premixed stream as far as I understand.

f1xpl, 2018-02-25:
You can try to test different resample methods (e.g. set those that do not use floats).

In my case below config did the job:

/etc/pulse/daemon.conf

default-fragments = 2
default-fragment-size-msec = 5
deferred-volume-safety-margin-usec = 1
flat-volumes = no
realtime-scheduling = no

Also you can be interested in PULSE_LATENCY_MSEC environment variable.

As you can see in [QAudioOutput](https://doc.qt.io/qt-5.10/qaudiooutput.html) documentation, there is nothing regarding remixing. It is quite simple class with only playback control and feeding methods.

Vitalic66, 2018-02-25:
Is it possible to read the output samplerate while stream is running?

Is it a fixed stream with 48000Hz or 44100Hz? Just read in the Qt doc about output of samplerate.

f1xpl, 2018-02-25:
Each audio stream has its separated QAudioOutput instance.

Media: 48KHz stereo
System: 16KHz mono
Speech: 16KHz mono

closed 2018-03-01 by f1xpl