f1xpl/openauto #23: Support for OEM-integrated devices

shawngmc, 2018-03-01:
Request: I can't find any documentation on whether or not openauto can integrate with existing OEM devices in vehicles, including speakers, voice control buttons, microphones, and cameras. Is there anything built-in to support these?

- I'm assuming the speakers and mic could be run to an appropriate USB sound card and be rather transparent.
- The buttons could be wired to GPIO or a USB input adapter like a Ultimarc IPAC; then, as long as they were mapped to valid keyboard shortcuts, that would also be fairly easy.
- The backup camera is the trickiest. Getting the video would presumably require the camera's output to be brought to composite, and then a USB Composite adapter could capture that video. But I'm not sure there's an easy way to bring that video to the screen, especially on-demand or when reversing.

Use Case: One powerful use case for openauto is to replace an existing head unit. As such, devices built into the vehicle - for example, in my vehicle, steering wheel buttons, headliner mic, rear view camera - would no longer be connected to anything. Standard Android Auto head units typically can support these.

f1xpl, 2018-03-01:
OpenAuto emulates part of head unit responsible for Android Auto. It does not support and will not support  in future (there is no such scope in the roadmap) specific car features like rear camera, FM/DAB radio or other functionalities unless they will be supported by Android Auto.

For now you must implement all specific car's head unit features by yourself.

shawngmc, 2018-03-01:
Ah, I see. That's a double edged sword, of course - on the one hand, it's not a drop-in thing, but the upside is that it can be more compatible/more focused on the core Android Auto functionality.

Given your response (and thank you for responding before I started HW shopping!), I'm perfectly fine with this being closed - but out of curiosity:
- Are there any pre-built RPi Head unit packages that OpenAuto works with/has been tried with? For example, iCarus?
- Are there any specific API for another process to control OpenAuto visibility, etc? 
- Alternatively, is OpenAuto more of a smoke test or example implementation of aasdk?

I'm just trying to figure out my level of effort to get the feature set I want.

f1xpl, 2018-03-01:
Basically OpenAuto does not implement any integration with 3rd part software.

OpenAuto is just a base for the AndroidAuto capable head unit. That was the main point - provide support of AndroidAuto on the RPI. There is no other car-specific features because it is almost imposible to implement universal solution for all cars. Almost all cars differs in CAN protocol, sensors and so on.

So just to sumarize. You can e.g. fork OpenAuto and design your own head unit software based on it. You can also use it as a standalone application to provide AndroidAuto support for your existing head unit software.

razvanioan, 2018-07-27:
Another question regarding custom reverse camera solution.

I am using an USB webcam, which I can display directly using mplayer
The only problem is that while phone is connected to OpenAuto, the video is under it and cannot be seen (until the phone get's disconnected)

Is there an option to change the video priority or disable from code / script the Android projection until the reverse operation has ended ?

Regards,

meekys, 2018-07-28:
I've written a little cam overlay app that might be of interest. It'll display a video stream from a v4l2 device, overlayed over other apps (including openauto)
https://github.com/meekys/cam_overlay
Just start the app when you want to show the reverse cam, kill the process when you no longer want to display anything.