f1xpl/openauto #24: Failed to run on x86

e8johan, 2018-03-02:
I'm running this on on Debian x86 system. I understand that this is not the intended hardware and probably will not work, but I still want to share the current state. This is the log that I get from connecting a device:

```
[2018-03-02 08:01:54.244048] [0x00007f6b71242700] [info]    [AaApp] [USBApp] Waiting for device...
UdevQt: unhandled device action "unbind"
UdevQt: unhandled device action "bind"
UdevQt: unhandled device action "unbind"
[2018-03-02 08:02:08.180953] [0x00007f6b6bfff700] [info]    [AaApp] [USBApp] Device connected.
[2018-03-02 08:02:08.182859] [0x00007f6b6bfff700] [info]    [AaApp] [AndroidAutoEntity] start.
[2018-03-02 08:02:08.183453] [0x00007f6b8bcded40] [debug]   [AaApp] [AudioInput] create.
UdevQt: unhandled device action "bind"
[2018-03-02 08:02:08.186226] [0x00007f6b8bcded40] [debug]   [AaApp] [AudioOutput] create.
[2018-03-02 08:02:08.186334] [0x00007f6b8bcded40] [debug]   [AaApp] [AudioOutput] create.
[2018-03-02 08:02:08.186393] [0x00007f6b8bcded40] [debug]   [AaApp] [AudioOutput] create.
[2018-03-02 08:02:08.186457] [0x00007f6b8bcded40] [debug]   [AaApp] [QtVideoOutput] create.
UdevQt: unhandled device action "bind"
[2018-03-02 08:02:08.256644] [0x00007f6b6bfff700] [info]    [AaApp] [AudioInputService] start.
[2018-03-02 08:02:08.259718] [0x00007f6b6bfff700] [info]    [AaApp] [AudioService] start, channel: MEDIA_AUDIO
[2018-03-02 08:02:08.260032] [0x00007f6b6bfff700] [info]    [AaApp] [AudioService] start, channel: SPEECH_AUDIO
[2018-03-02 08:02:08.260155] [0x00007f6b6bfff700] [info]    [AaApp] [AudioService] start, channel: SYSTEM_AUDIO
[2018-03-02 08:02:08.260230] [0x00007f6b6bfff700] [info]    [AaApp] [SensorService] start.
[2018-03-02 08:02:08.260479] [0x00007f6b6bfff700] [info]    [AaApp] [VideoService] start.
[2018-03-02 08:02:08.260728] [0x00007f6b6bfff700] [info]    [AaApp] [BluetoothService] start.
[2018-03-02 08:02:08.260986] [0x00007f6b6bfff700] [info]    [AaApp] [InputService] start.
[2018-03-02 08:02:09.061903] [0x00007f6b70a41700] [info]    [AaApp] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-03-02 08:02:09.061956] [0x00007f6b70a41700] [info]    [AaApp] [AndroidAutoEntity] Begin handshake.
[2018-03-02 08:02:09.087617] [0x00007f6b70a41700] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 2282
[2018-03-02 08:02:09.096124] [0x00007f6b70a41700] [info]    [AaApp] [AndroidAutoEntity] continue handshake.
[2018-03-02 08:02:09.119933] [0x00007f6b70a41700] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 51
[2018-03-02 08:02:09.120411] [0x00007f6b70a41700] [info]    [AaApp] [AndroidAutoEntity] Auth completed.
[2018-03-02 08:02:09.218375] [0x00007f6b71242700] [info]    [AaApp] [AndroidAutoEntity] Discovery request, device name: Android, brand: samsung SM-G930F
[2018-03-02 08:02:09.218462] [0x00007f6b71242700] [info]    [AaApp] [AudioInputService] fill features.
[2018-03-02 08:02:09.218530] [0x00007f6b71242700] [info]    [AaApp] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-03-02 08:02:09.218570] [0x00007f6b71242700] [info]    [AaApp] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-03-02 08:02:09.218625] [0x00007f6b71242700] [info]    [AaApp] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-03-02 08:02:09.218658] [0x00007f6b71242700] [info]    [AaApp] [SensorService] fill features.
[2018-03-02 08:02:09.218691] [0x00007f6b71242700] [info]    [AaApp] [VideoService] fill features.
[2018-03-02 08:02:09.218753] [0x00007f6b71242700] [info]    [AaApp] [BluetoothService] fill features
[2018-03-02 08:02:09.218784] [0x00007f6b71242700] [info]    [AaApp] [InputService] fill features.
[2018-03-02 08:02:09.476669] [0x00007f6b6bfff700] [info]    [AaApp] [SensorService] open request, priority: 0
[2018-03-02 08:02:09.476731] [0x00007f6b6bfff700] [info]    [AaApp] [SensorService] open status: 0
[2018-03-02 08:02:09.516481] [0x00007f6b6bfff700] [info]    [AaApp] [SensorService] sensor start request, type: 13
[2018-03-02 08:02:09.778826] [0x00007f6b71242700] [info]    [AaApp] [AndroidAutoEntity] Shutdown request, reason: 1
[2018-03-02 08:02:09.779236] [0x00007f6b6b7fe700] [info]    [AaApp] [USBApp] quit.
[2018-03-02 08:02:09.779257] [0x00007f6b6b7fe700] [info]    [AaApp] [AndroidAutoEntity] stop.
[2018-03-02 08:02:09.779275] [0x00007f6b6b7fe700] [info]    [AaApp] [AudioInputService] stop.
[2018-03-02 08:02:09.779314] [0x00007f6b6b7fe700] [info]    [AaApp] [AudioService] stop, channel: MEDIA_AUDIO
[2018-03-02 08:02:09.779335] [0x00007f6b6b7fe700] [info]    [AaApp] [AudioService] stop, channel: SPEECH_AUDIO
[2018-03-02 08:02:09.779352] [0x00007f6b6b7fe700] [info]    [AaApp] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-03-02 08:02:09.779370] [0x00007f6b6b7fe700] [info]    [AaApp] [SensorService] stop.
[2018-03-02 08:02:09.779387] [0x00007f6b6b7fe700] [info]    [AaApp] [VideoService] stop.
[2018-03-02 08:02:09.779406] [0x00007f6b6b7fe700] [info]    [AaApp] [BluetoothService] stop.
[2018-03-02 08:02:09.779423] [0x00007f6b6b7fe700] [info]    [AaApp] [InputService] stop.
[2018-03-02 08:02:09.779435] [0x00007f6b6b7fe700] [info]    [AaApp] [InputDevice] stop.
[2018-03-02 08:02:09.779667] [0x00007f6b6bfff700] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 31, native code: 0
[2018-03-02 08:02:09.779843] [0x00007f6b71242700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 31, native code: 0, channel: MEDIA_AUDIO
[2018-03-02 08:02:09.779866] [0x00007f6b6bfff700] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 31, native code: 0
[2018-03-02 08:02:09.779915] [0x00007f6b6bfff700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 31, native code: 0, channel: SYSTEM_AUDIO
[2018-03-02 08:02:09.779934] [0x00007f6b70a41700] [error]   [AaApp] [BluetoothService] channel error: AaSdk error code: 31, native code: 0
[2018-03-02 08:02:09.779957] [0x00007f6b71242700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 31, native code: 0, channel: SPEECH_AUDIO
[2018-03-02 08:02:09.779971] [0x00007f6b6bfff700] [error]   [AaApp] [VideoService] channel error: AaSdk error code: 31, native code: 0
[2018-03-02 08:02:09.780127] [0x00007f6b6b7fe700] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-02 08:02:09.780245] [0x00007f6b70a41700] [error]   [AaApp] [AudioInputService] channel error: AaSdk error code: 31, native code: 0
[2018-03-02 08:02:09.780275] [0x00007f6b6b7fe700] [error]   [AaApp] [AndroidAutoEntity] channel error: AaSdk error code: 31, native code: 0
[2018-03-02 08:02:09.780292] [0x00007f6b6b7fe700] [debug]   [AaApp] [AndroidAutoEntity] destroy.
UdevQt: unhandled device action "unbind"
UdevQt: unhandled device action "unbind"
```

Is this expected? What would be the next steps? Is there anything that I can help with?

f1xpl, 2018-03-02:
It would be good to collect adb logs. Also please provide you smartphone model, Android OS version, AndroidAuto app version.

e8johan, 2018-03-02:
It is a Galaxy S7 (Samsung SM-G930F), running Android 7.0. I use 640c0c6dc9a02e9c8a3c1609624696fb2c321aad of openandroid and 5edcc7c68c0a8ec27734a008fb3b45254b996910 of aasdk

e8johan, 2018-03-02:
How do I collect the adb logs?

parkerlreed, 2018-03-02:
Did you make sure to set the RPI3 flag to false?

`-DRPI3_BUILD=FALSE`

I've had it running on multiple x86_64 machines without issue.

e8johan, 2018-03-03:
Just tried a clean build with this, but no change in the behaviour

```
[2018-03-03 20:18:32.851116] [0x00007f4e9ca05700] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-03 20:18:43.800742] [0x00007f4e97fff700] [info]    [AaApp] [USBApp] Device connected.
[2018-03-03 20:18:43.808148] [0x00007f4e97fff700] [info]    [AaApp] [AndroidAutoEntity] start.
[2018-03-03 20:18:43.809936] [0x00007f4eb7439d40] [debug]   [AaApp] [AudioInput] create.
[2018-03-03 20:18:43.815478] [0x00007f4eb7439d40] [debug]   [AaApp] [AudioOutput] create.
[2018-03-03 20:18:43.815636] [0x00007f4eb7439d40] [debug]   [AaApp] [AudioOutput] create.
[2018-03-03 20:18:43.815706] [0x00007f4eb7439d40] [debug]   [AaApp] [AudioOutput] create.
[2018-03-03 20:18:43.815822] [0x00007f4eb7439d40] [debug]   [AaApp] [QtVideoOutput] create.
[2018-03-03 20:18:43.944953] [0x00007f4e97fff700] [info]    [AaApp] [AudioInputService] start.
[2018-03-03 20:18:43.946598] [0x00007f4e97fff700] [info]    [AaApp] [AudioService] start, channel: MEDIA_AUDIO
[2018-03-03 20:18:43.946770] [0x00007f4e97fff700] [info]    [AaApp] [AudioService] start, channel: SPEECH_AUDIO
[2018-03-03 20:18:43.946837] [0x00007f4e97fff700] [info]    [AaApp] [AudioService] start, channel: SYSTEM_AUDIO
[2018-03-03 20:18:43.946879] [0x00007f4e97fff700] [info]    [AaApp] [SensorService] start.
[2018-03-03 20:18:43.947019] [0x00007f4e97fff700] [info]    [AaApp] [VideoService] start.
[2018-03-03 20:18:43.947157] [0x00007f4e97fff700] [info]    [AaApp] [BluetoothService] start.
[2018-03-03 20:18:43.947297] [0x00007f4e97fff700] [info]    [AaApp] [InputService] start.
[2018-03-03 20:18:45.184221] [0x00007f4e9ca05700] [info]    [AaApp] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-03-03 20:18:45.184275] [0x00007f4e9ca05700] [info]    [AaApp] [AndroidAutoEntity] Begin handshake.
[2018-03-03 20:18:45.214588] [0x00007f4e97fff700] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 2282
[2018-03-03 20:18:45.223837] [0x00007f4e97fff700] [info]    [AaApp] [AndroidAutoEntity] continue handshake.
[2018-03-03 20:18:45.254915] [0x00007f4e9ca05700] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 51
[2018-03-03 20:18:45.255407] [0x00007f4e9ca05700] [info]    [AaApp] [AndroidAutoEntity] Auth completed.
[2018-03-03 20:18:45.346297] [0x00007f4e96ffd700] [info]    [AaApp] [AndroidAutoEntity] Discovery request, device name: Android, brand: samsung SM-G930F
[2018-03-03 20:18:45.346407] [0x00007f4e96ffd700] [info]    [AaApp] [AudioInputService] fill features.
[2018-03-03 20:18:45.346476] [0x00007f4e96ffd700] [info]    [AaApp] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-03-03 20:18:45.346520] [0x00007f4e96ffd700] [info]    [AaApp] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-03-03 20:18:45.346550] [0x00007f4e96ffd700] [info]    [AaApp] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-03-03 20:18:45.346580] [0x00007f4e96ffd700] [info]    [AaApp] [SensorService] fill features.
[2018-03-03 20:18:45.346616] [0x00007f4e96ffd700] [info]    [AaApp] [VideoService] fill features.
[2018-03-03 20:18:45.346651] [0x00007f4e96ffd700] [info]    [AaApp] [BluetoothService] fill features
[2018-03-03 20:18:45.346678] [0x00007f4e96ffd700] [info]    [AaApp] [InputService] fill features.
[2018-03-03 20:18:45.433054] [0x00007f4e96ffd700] [info]    [AaApp] [SensorService] open request, priority: 0
[2018-03-03 20:18:45.433112] [0x00007f4e96ffd700] [info]    [AaApp] [SensorService] open status: 0
[2018-03-03 20:18:45.449359] [0x00007f4e977fe700] [info]    [AaApp] [SensorService] sensor start request, type: 13
[2018-03-03 20:18:45.711254] [0x00007f4e96ffd700] [info]    [AaApp] [AndroidAutoEntity] Shutdown request, reason: 1
[2018-03-03 20:18:45.711906] [0x00007f4e977fe700] [info]    [AaApp] [USBApp] quit.
[2018-03-03 20:18:45.711955] [0x00007f4e977fe700] [info]    [AaApp] [AndroidAutoEntity] stop.
[2018-03-03 20:18:45.711986] [0x00007f4e977fe700] [info]    [AaApp] [AudioInputService] stop.
[2018-03-03 20:18:45.712061] [0x00007f4e977fe700] [info]    [AaApp] [AudioService] stop, channel: MEDIA_AUDIO
[2018-03-03 20:18:45.712100] [0x00007f4e977fe700] [info]    [AaApp] [AudioService] stop, channel: SPEECH_AUDIO
[2018-03-03 20:18:45.712128] [0x00007f4e977fe700] [info]    [AaApp] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-03-03 20:18:45.712158] [0x00007f4e977fe700] [info]    [AaApp] [SensorService] stop.
[2018-03-03 20:18:45.712183] [0x00007f4e977fe700] [info]    [AaApp] [VideoService] stop.
[2018-03-03 20:18:45.712224] [0x00007f4e977fe700] [info]    [AaApp] [BluetoothService] stop.
[2018-03-03 20:18:45.712255] [0x00007f4e977fe700] [info]    [AaApp] [InputService] stop.
[2018-03-03 20:18:45.712273] [0x00007f4e977fe700] [info]    [AaApp] [InputDevice] stop.
[2018-03-03 20:18:45.712714] [0x00007f4e9ca05700] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 31, native code: 0
[2018-03-03 20:18:45.712757] [0x00007f4e97fff700] [error]   [AaApp] [BluetoothService] channel error: AaSdk error code: 31, native code: 0
[2018-03-03 20:18:45.712799] [0x00007f4e96ffd700] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 31, native code: 0
[2018-03-03 20:18:45.712850] [0x00007f4e9ca05700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 31, native code: 0, channel: MEDIA_AUDIO
[... cut ...]
[2018-03-03 20:18:45.712974] [0x00007f4e97fff700] [error]   [AaApp] [AudioInputService] channel error: AaSdk error code: 31, native code: 0
[2018-03-03 20:18:45.713396] [0x00007f4e977fe700] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-03 20:18:45.713540] [0x00007f4e977fe700] [error]   [AaApp] [AndroidAutoEntity] channel error: AaSdk error code: 32, native code: 0
[2018-03-03 20:18:45.713575] [0x00007f4e977fe700] [debug]   [AaApp] [AndroidAutoEntity] destroy.
```

I guess the next step is the adb logs, but I'm not sure how to capture them. Do you have a pointer to where I can read up on that?

e8johan, 2018-03-03:
So, now I figured out how to grab adb logs. There is a ton of stuff here. What am I looking for?

e8johan, 2018-03-03:
I googled around a bit and this seems to be a part of the issue: https://productforums.google.com/forum/#!topic/android-auto/kaTiOBRilw0;context-place=topicsearchin/android-auto/galaxy$20s7$20android$20incompatible$20device

Now my phone warns that I'm running an incompatible version (I'm running AA 3.0.570554-release-30570554).

f1xpl, 2018-03-04:
Please checl audio and video capabilities of your system (audio playback up to 48KHz and present of H.264 codec).

closed 2018-03-06 by f1xpl