f1xpl/openauto #25: MacOS support

duxet, 2018-03-02:
There are some issues like:
- [ ] protobuf error: File already exists in database: AVMediaAckIndicationMessage.proto
- [ ] missing video (black screen)

duxet, 2018-03-06:
Video is working fine with the same device on Linux platform, so definitely this is Mac OS issue.

f1xpl, 2018-03-06:
I think the most suspect are video codecs.

f1xpl, 2018-03-10:
Could you please retest compilation on MacOS with the latest master?

duxet, 2018-03-12:
```
[2018-03-12 09:26:11.253791] [0x00007fffd16363c0] [warning] [AaApp] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
[2018-03-12 09:26:11.521941] [0x0000700006aba000] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-12 09:26:16.570022] [0x0000700006bc0000] [info]    [AaApp] [USBApp] Device connected.
[2018-03-12 09:26:16.573025] [0x0000700006bc0000] [info]    [AaApp] [AndroidAutoEntity] start.
[2018-03-12 09:26:16.573591] [0x00007fffd16363c0] [debug]   [AaApp] [AudioInput] create.
[2018-03-12 09:26:16.594238] [0x00007fffd16363c0] [debug]   [AaApp] [AudioOutput] create.
[2018-03-12 09:26:16.594928] [0x00007fffd16363c0] [debug]   [AaApp] [AudioOutput] create.
[2018-03-12 09:26:16.595482] [0x00007fffd16363c0] [debug]   [AaApp] [AudioOutput] create.
[2018-03-12 09:26:16.595979] [0x00007fffd16363c0] [debug]   [AaApp] [QtVideoOutput] create.
[2018-03-12 09:26:16.601730] [0x0000700006bc0000] [info]    [AaApp] [AudioInputService] start.
[2018-03-12 09:26:16.601846] [0x0000700006bc0000] [info]    [AaApp] [AudioService] start, channel: MEDIA_AUDIO
[2018-03-12 09:26:16.601859] [0x0000700006bc0000] [info]    [AaApp] [AudioService] start, channel: SPEECH_AUDIO
[2018-03-12 09:26:16.601868] [0x0000700006bc0000] [info]    [AaApp] [AudioService] start, channel: SYSTEM_AUDIO
[2018-03-12 09:26:16.601875] [0x0000700006bc0000] [info]    [AaApp] [SensorService] start.
[2018-03-12 09:26:16.601883] [0x0000700006bc0000] [info]    [AaApp] [VideoService] start.
[2018-03-12 09:26:16.601892] [0x0000700006bc0000] [info]    [AaApp] [BluetoothService] start.
[2018-03-12 09:26:16.601911] [0x0000700006bc0000] [info]    [AaApp] [InputService] start.
[2018-03-12 09:26:16.630364] [0x0000700006b3d000] [info]    [AaApp] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-03-12 09:26:16.630396] [0x0000700006b3d000] [info]    [AaApp] [AndroidAutoEntity] Begin handshake.
[2018-03-12 09:26:16.641224] [0x0000700006aba000] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 2311
[2018-03-12 09:26:16.645616] [0x0000700006aba000] [info]    [AaApp] [AndroidAutoEntity] continue handshake.
[2018-03-12 09:26:16.652858] [0x0000700006b3d000] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 51
[2018-03-12 09:26:16.653012] [0x0000700006b3d000] [info]    [AaApp] [AndroidAutoEntity] Auth completed.
[2018-03-12 09:26:16.673234] [0x0000700006b3d000] [info]    [AaApp] [AndroidAutoEntity] Discovery request, device name: Android, brand: OnePlus ONEPLUS A5000
[2018-03-12 09:26:16.673307] [0x0000700006b3d000] [info]    [AaApp] [AudioInputService] fill features.
[2018-03-12 09:26:16.673340] [0x0000700006b3d000] [info]    [AaApp] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-03-12 09:26:16.673355] [0x0000700006b3d000] [info]    [AaApp] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-03-12 09:26:16.673367] [0x0000700006b3d000] [info]    [AaApp] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-03-12 09:26:16.673379] [0x0000700006b3d000] [info]    [AaApp] [SensorService] fill features.
[2018-03-12 09:26:16.673392] [0x0000700006b3d000] [info]    [AaApp] [VideoService] fill features.
[2018-03-12 09:26:16.673404] [0x0000700006b3d000] [info]    [AaApp] [BluetoothService] fill features
[2018-03-12 09:26:16.673415] [0x0000700006b3d000] [info]    [AaApp] [InputService] fill features.
[2018-03-12 09:26:16.710661] [0x0000700006b3d000] [info]    [AaApp] [AudioInputService] open request, priority: 0
[2018-03-12 09:26:16.710692] [0x0000700006b3d000] [info]    [AaApp] [AudioInputService] open status: 0
[2018-03-12 09:26:16.711228] [0x0000700006a37000] [info]    [AaApp] [AndroidAutoEntity] requested audio focus, type: 4
[2018-03-12 09:26:16.711247] [0x0000700006a37000] [info]    [AaApp] [AndroidAutoEntity] audio focus state: 3
[2018-03-12 09:26:16.712348] [0x0000700006a37000] [info]    [AaApp] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-03-12 09:26:16.712387] [0x0000700006a37000] [info]    [AaApp] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2018-03-12 09:26:16.712741] [0x0000700006bc0000] [info]    [AaApp] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2018-03-12 09:26:16.712769] [0x0000700006bc0000] [info]    [AaApp] [AudioService] open status: 0, channel: SPEECH_AUDIO
[2018-03-12 09:26:16.713038] [0x0000700006b3d000] [info]    [AaApp] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2018-03-12 09:26:16.713059] [0x0000700006b3d000] [info]    [AaApp] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[2018-03-12 09:26:16.713327] [0x0000700006aba000] [info]    [AaApp] [SensorService] open request, priority: 0
[2018-03-12 09:26:16.713344] [0x0000700006aba000] [info]    [AaApp] [SensorService] open status: 0
[2018-03-12 09:26:16.713582] [0x0000700006b3d000] [info]    [AaApp] [VideoService] open request, priority: 0
[2018-03-12 09:26:16.713603] [0x0000700006b3d000] [debug]   [AaApp] [QtVideoOutput] open.
[2018-03-12 09:26:16.713616] [0x0000700006b3d000] [info]    [AaApp] [VideoService] open status: 0
[2018-03-12 09:26:16.713880] [0x0000700006b3d000] [info]    [AaApp] [InputService] open request, priority: 0
[2018-03-12 09:26:16.713897] [0x0000700006b3d000] [info]    [AaApp] [InputService] open status: 0
[2018-03-12 09:26:16.717927] [0x0000700006aba000] [info]    [AaApp] [VideoService] setup request, config index: 3
[... cut ...]
[2018-03-12 09:26:28.242816] [0x0000700006b3d000] [info]    [AaApp] [InputDevice] stop.
[2018-03-12 09:26:28.243028] [0x0000700006b3d000] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-12 09:26:28.243053] [0x0000700006b3d000] [debug]   [AaApp] [AndroidAutoEntity] destroy.
```

i've added some debug stuff to QtVideoOutput, but seems like the media player status is `QMediaPlayer::NoMedia`, which means `The is no current media. The player is in the StoppedState.` and there are no calls on error and mediaStatusChanged events

f1xpl, 2018-03-12:
There is a remark in [mediaStatus method](https://doc.qt.io/qt-5.10/qmediaplayer.html#mediaStatus-prop)

> By default this property is QMediaPlayer::NoMedia

I think the value of state() would be more helpful in investigation.

I am wondering about "Stream source" [in this document](http://wiki.qt.io/Qt_5.7_Multimedia_Backends). It is not checked for OS X.

duxet, 2018-03-12:
The state code is `QMediaPlayer::StoppedState`. Probably missing stream source is a reason. I was thinking about using gstreamer backend instead of default one on Mac OS, but i have no idea how to do that easily.

cupral, 2018-04-08:
I have same black screen on Ubuntu on VirtualBox on OSX

cheriff, 2018-07-22:
If I hack QtVideoOutput::write() to dump bytes to a file (in addition to the videobuffer) I do end up with a file that is playable, and openauto displays the abovementioned black screen.

```
$ mplayer media.h264  -fps 30
MPlayer 1.3.0-4.2.1 (C) 2000-2016 MPlayer Team
Can't init Apple Remote.

Playing media.h264.
libavformat version 57.25.100 (internal)
H264-ES file format detected.
2018-07-22 22:34:47.464 mplayer[59589:6342513] *** WARNING: Textured window <NSWindow: 0x7f8854c3a740> is getting an implicitly transparent titlebar. This will break when linking against newer SDKs. Use NSWindow's -titlebarAppearsTransparent=YES instead.
==========================================================================
Opening video decoder: [ffmpeg] FFmpeg's libavcodec codec family
libavcodec version 57.24.102 (internal)
Selected video codec: [ffh264] vfm: ffmpeg (FFmpeg H.264)
==========================================================================
Load subtitles in ./
Audio: no sound
FPS forced to be 30.000  (ftime: 0.033).
Starting playback...
Could not find matching colorspace - retrying with -vf scale...
Opening video filter: [scale]
Movie-Aspect is undefined - no prescaling applied.
[swscaler @ 0x10b66f970]bicubic scaler, from yuv420p to yuyv422 using MMXEXT
VO: [corevideo] 800x480 => 800x480 Packed YUY2
[ASPECT] Warning: No suitable new res found!
V:   0.0 127/127  5%  6%  0.0% 0 0

Exiting... (End of file)
```

I'll note:
1. The -fps 30 is required, else lots of decoding errors and nothing displays. The value doesn't seem to matter, other than being fast/slow-mo if incorrect
2. VLC also doesn't like the file.

I don't know much about the QT video player and codec support ... is this at all useful info?
Am happy to post the ~420K file if requested.

f1xpl, 2018-07-22:
@cheriff According to the Qt documentation, video backend used for MacOS does not support video streaming feature.

cheriff, 2018-07-22:
@f1xpl ah ok, I guess that explains that, then :)