f1xpl/openauto #26: "Waiting for device" screen enhancements

htruong, 2018-03-03:
I'm thinking of something like this, so the interface isn't cluttered with warnings: I think people are aware of the consequences of using OpenAuto.

Every button has an ample hit target, so using with a fat finger is easy.

I'm also proposing a new button to turn the screen off (the car "sleep" button) so it doesn't keep the screen on and thus doesn't attract thieves. To turn the screen back on, the user would have to tap anywhere on the screen or connect the phone to the head unit.

Here is a sketch:

[image]

f1xpl, 2018-03-03:
Idea is great but in my opinion disclaimer screen should be kept. I think not all OpenAuto's users (especially those without programming skills) are aware that OpenAuto is just AndroidAuto emulator and has nothing to do with head units certified by Google. OpenAuto could introduce some bugs that may not be present on official head units. Those bugs may affect reputation of AndroidAuto.

htruong, 2018-03-03:
I see your reasoning, I think that's a good point -- we don't want to upset Google. I wonder if you're open to the idea of having a toast message that the user has to see for a minute or two that overlays on the main screen. That way, the user doesn't always have to see it, and at the same time, it can be aesthetically pleasing to the eyes.

htruong, 2018-03-31:
OK, I have a better "plug phone in" screen and is almost ready to get a pull request. The iconography on the left is not displayed here, but the user can choose to put their own wallpaper to get rid of the message to make it whatever they want if they already read the default warning. What do you think?

[image]

quattrofoto, 2018-07-18:
I'd like to keep the stock layout and disclaimers and information but is it possible to customise the background behind it - ie. instead of the grey background, have an image file (for instance I have made a custom Land Rover image for my first build, as it is for my classic 1963 Land Rover, and I've just got another one up and running for my Ford boring so I'd like to have a Ford themed background for that one)

However I haven't been able to find any way to change the background at all?

WarDrake, 2018-07-19:
@quattrofoto it is pretty easy actually, you need to have a file you want yo use as background, add it to the resources file inside assets and then on the mainscreen.ui file add a label that's behind everything and invokes the file you put in assets, and then re compile OA, play around with the code a little and you'll see where everything is and how easy it is to mod something.

quattrofoto, 2018-07-19:
You lost me at "add it to the resources file"...  Sorry I'm not a complete novice, but I have no idea how to do what you advise!  I can get to the mainscreen.ui file ok but I don't know what the syntax is to do what you suggest? Also, I wouldn't know how to recompile OA?  Apologies if I sound thick, I did actually do a bit of Java programming at Uni but that's over ten years ago now and I haven't used it since.... :-(