f1xpl/openauto #27: physical button for the mic on the raspberry pi

keredson, 2018-03-04:
openauto may support this already, but i couldn't find any references to it.  :)

openauto should support manually triggering the mic (for voice recognition) via one of the open pins on the raspberry pi.  this way we could wire up a (NO) button to act as the green voice button on a normal andriod auto.

it should use one of the pins with an internal pull-up resistor resistor so we could just short it directly to ground.

we should listen to the interrupt (to avoid polling).

thanks!

f1xpl, 2018-03-04:
You can use keyboard to trigger mic (voice assistant) and map that keyboard event to the desired GPIO pin state.

keredson, 2018-03-05:
is that a feature of openauto, or are you just stating in general?  (i found a multitude of scripts online that claim to do that, but no real consensus.  mostly one-offs for individual projects.)

f1xpl, 2018-03-05:
OpenAuto supports keyboard input - [here](https://github.com/f1xpl/openauto/wiki/Keyboard-button-bindings) is the list of supported bindings. Stuff related to the GPIO tiggers you must do by yourself. I think it can be easily done using python.

closed 2018-03-06 by f1xpl