f1xpl/openauto #28: [Enhancement] Media key handling

parkerlreed, 2018-03-06:
For people connecting their car's Bluetooth to the computer running OpenAuto, this lets some use the built-in buttons to control the music.

In https://github.com/f1xpl/openauto/blob/development/src/autoapp/Projection/InputDevice.cpp

````
--- /home/parker/build/InputDevice.cpp	2018-03-05 23:28:06.828867631 -0500
+++ InputDevice.cpp	2018-03-05 23:27:53.036062532 -0500
@@ -133,14 +133,17 @@
         buttonCode = aasdk::proto::enums::ButtonCode::PAUSE;
         break;
 
+    case Qt::Key_MediaPrevious:
     case Qt::Key_V:
         buttonCode = aasdk::proto::enums::ButtonCode::PREV;
         break;
 
+    case Qt::Key_MediaPlay:
     case Qt::Key_B:
         buttonCode = aasdk::proto::enums::ButtonCode::TOGGLE_PLAY;
         break;
 
+    case Qt::Key_MediaNext:
     case Qt::Key_N:
         buttonCode = aasdk::proto::enums::ButtonCode::NEXT;
         break;
````

The Setting screen help text may need to be updated to reflect the changes.

f1xpl, 2018-03-06:
Could you please create pull request for this change? Maybe it would be better to use Qt::Key_MediaPlay as toggle play button?

f1xpl, 2018-03-06:
Done 1af06dae160d5e3c42b0a2d3e4d9e94cced31571

closed 2018-03-06 by f1xpl