f1xpl/openauto #30: Vehicle data display

duxet, 2018-03-06:
I was wondering: is it feasible to display in Android Auto things like engine RPM or HVAC settings?

From what i was able to gather from my little research seems like there is a possibility to send some vehicle data from HU to AA [1]. But is it connected somehow to Android "Car API" [2]?

There is even some existing app to do that [3], but could it be run on AA after applying some modifications?

I tried to find answers to these questions on my own, but this whole stuff seems to be huge and separated into different places: core Android project, GMS and AA app itself...

[1] https://github.com/f1xpl/aasdk/blob/development/aasdk_proto/HVACData.proto
[2] https://android.googlesource.com/platform/packages/services/Car/+/nougat-dev/car-lib/src/android/car/hardware/hvac/CarHvacManager.java
[3] https://android.googlesource.com/platform/packages/apps/Car/Hvac/+/master

hitech95, 2018-03-14:
I'm not an expert of AA but if I got this right:
To make this possible you need a special sdk made by google, or you can extract it from other apps and build your own app.

Headunit side doesn't support apps. But you have to write a software able to interface via the automotive hal to an android device.

duxet, 2018-03-19:
Seems like #33 could be a great opportunity to add support for sync of vehicle data using CAN bus from head unit to AA.

kire73, 2018-03-28:
There's an easier way if you're willing to work out a few extra bucks for an ODB2 reader and compatible app. There is an app called Torque pro for ODB readers that is very customizeable and extensively loaded with features. It was denied by Google for AA compatibility, but the developer made it unlockable.

You would need the Pro version of the app, as well as the XDA Labs Dev app, where you can get the plug-in ODB2AA. This makes your torque screens visible in AA.

https://forum.xda-developers.com/general/paid-software/app-obd2-plugin-android-auto-torque-t3657805

Looks like it's temporarily unavailable until a pending fix is issued at time of post.