f1xpl/openauto #32: Rockchip SoC support

synssins, 2018-03-06:
I'm not sure where to put this as there isn't a forum for OpenAuto that I have yet found.

The Tinker Board uses the RockChip SoC, so I run into issues with the Broadcom and Raspberry Pi Firmware stage of the build. I have a Pi3b I can use, but was hoping to put the better DAC of the Tinker Board into play.

f1xpl, 2018-03-06:
Remove -DRPI3_BUILD=TRUE from cmake command. It will disable building of OMX video output.

synssins, 2018-03-06:
Thank you. I was able to successfully build with no issues.
When I run autoapp, I get the following:

`[2018-03-06 22:18:19.045430] [0xb1e1d060] [warning] [AaApp] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file
QXcbConnection: Could not connect to display
Aborted
`

synssins, 2018-03-06:
I think I got it. Followed the Autostart instructions, rebooted the device and it loaded immediately.
Now to test.

parkerlreed, 2018-03-06:
Yeah the non RPi3 build is expecting X11 to be running.

synssins, 2018-03-06:
Yeah. x11 was running, but I was SSH'd in from my Windows PC trying to launch it. I set it to autorun on start and it worked.
The issue I ran into after that was that the display would go pure white, though everything still functioned from a touch perspective. Some bugs to work through on Non-Pi systems with rendering the display.

I'm tossing it on my Pi3 now.

closed 2018-03-07 by synssins