f1xpl/openauto #33: Plugin architecture

htruong, 2018-03-08:
There are certain tasks that would otherwise be quite hairy and architecture/environment-dependent if handled directly by OpenAuto. For example: dimming the screen, shutting down the OS, putting the display to sleep, handling Bluetooth, handling/displaying car sensors data, backup cameras... 

I wonder if that's possible to architect a plugin system to handle those needs.

synssins, 2018-03-08:
One solution that I'd like to see implemented is ignition controlled shutdown. Using a GPIO pin to detect when the ignition is on or off would control power state.

WhateverFits, 2018-03-08:
OBDII will give that kind of state so this is where a plug-in style
architecture would be useful. Using a pin or using OBDII to handle the
car's state could then affect the system state.

LiamLyt, 2018-03-24:
Carloop.io has a pretty good device that I feel wouldn't be too hard to merge with Open Auto

leaskovski, 2018-03-29:
This sort of functionality would be good.  I wouldn't mind somehow linking the KBus of a BMW to this so that I can use the idrive nob to navigate around AA.

f1xpl, 2018-03-29:
I am using iDrive from E60 in my OpenAuto setup. Plugin feature is not needed to make it work. You can simulate keyboard presses in order to control AA using custom input device (e.g. iDrive controller).

leaskovski, 2018-03-30:
Please tell me more.  What board have you bought to hook in with the PI to achive this.  Can you document this somewhere?  Many thanks!

martenumberto, 2018-04-19:
An integration of the rtl-sdr project would be awesome. With a cheap dvbt stick you can receive fm/DAB/... There are a lots of programs for Linux and also for Android.

retrorepair, 2018-05-03:
Has to be said, adding fm/dab would convince me to ditch my OEM head unit. It's the only thing that would be missing I think.

Glider95, 2018-07-02:
Hey !

Any News for all those ideas ? (OBD2 , FM RTL-SDR usb stick, ignition detection,backup camera ...) :)