f1xpl/openauto #37: Multi-Touch support

ginkage, 2018-03-10:
It would be nice to enhance the current input event handling to not only support QEvent::Mouse* events, but also QEvent::Touch* ones.
At a glance, it seems that this would require adding some sort of a toggle (allowing either mouse, or touch, but not both), that would define the InputDevice::eventFilter behaviour.
Then, if the touch is enabled (instead of mouse), some new routine would loop through QTouchEvent->touchPoints() and send proper eventHandler_->onTouchEvent calls for every pointer that's not "Qt::TouchPointStationary".
Seems pretty straightforward, and would greatly improve map zooming. :)

f1xpl, 2018-03-10:
Hello,
I can be wrong buit I have not seen any Head Unit with multitouch in action. Do you have any link to the video how it works?

ginkage, 2018-03-10:
Hi!I don't have any video, but this article:https://arstechnica.com/cars/2015/07/android-auto-review-a-beautiful-but-beta-alternative-to-awful-oem-solutions/2/#h2does mention multi-touch support in Hyundai Sonata unit. Might be worth a try anyway.

  
  

Kind%20regards%2C%0AIvan

ginkage, 2018-03-10:
Well, I tried to do the thing I've described in the first post, and propagated multiple-pointer events as is, replacing hard-coded "0" with multiple pointer IDs (not forgetting to change hard-coded value in InputService::onTouchEvent as well), but it didn't seem to work. Could be just the Android Auto interface limitation (i.e. the app itself not supporting multi-touch, even if the platform itself supports it), but oh well.
On the upside, all usual touch functions still worked fine, even though I removed the mouse events completely in favor of the touch events.

JMonsorno, 2018-03-14:
The Ford's have multi-touch, I even found a video. Although they don't do a good job demonstrating pinch to zoom, they get it and mess up a few times which happens to show another multi-touch gesture as well, rotation.
https://youtu.be/DRsC4xJbDvk?t=4m50s

JLOrwell, 2018-03-17:
I have the pioneer avh4100nex deck.  It has multi-touch.  It's required for certification so all android auto branded decks are supposed to be multi touch.

borconi, 2018-06-11:
fixpl you're wrong, I have added multitouch on my latest Android version. 

Here are relevant bits of code (Java):

```
 public void touch_send(byte action, long ts, CoordinatesHelpers... coordArray)
  {

    int actionindex=coordArray[0].actionIndex;

    byte[] ba_touch3 = new byte[]{AA_CH_TOU, 0x0b, 0x00, 0x00, -128, 0x01, 0x08} ;//0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0, 0};
    Input.TouchEvent.Builder touchevent = Input.TouchEvent.newBuilder();
    touchevent.setAction(Input.TouchEvent.PointerAction.forNumber(action));
    touchevent.setActionIndex(actionindex);
    for (CoordinatesHelpers currCordinate:coordArray) {
      Input.TouchEvent.Pointer.Builder pointerdata = Input.TouchEvent.Pointer.newBuilder();
      pointerdata.setX(currCordinate.x);
      pointerdata.setY(currCordinate.y);
      pointerdata.setPointerId(currCordinate.id);
      touchevent.addPointerData(pointerdata);
    }
    Input.TouchEvent builtdata = touchevent.build();
    builtdata.getSerializedSize();
    ByteArrayOutputStream tempStream = new ByteArrayOutputStream();
    CodedOutputStream codedStream = CodedOutputStream.newInstance(tempStream);

    try {
      codedStream.writeRawBytes(ByteBuffer.allocate(4).putInt( builtdata.getSerializedSize()+9+CodedOutputStream.computeUInt64SizeNoTag(ts)).array());
      codedStream.writeRawBytes(ba_touch3);
      codedStream.writeUInt64NoTag(ts);
      codedStream.writeRawByte(0x1a);
      codedStream.writeRawByte(builtdata.getSerializedSize());
      codedStream.writeRawBytes(builtdata.toByteArray());
      codedStream.flush();
      tempStream.writeTo(send_que);
    } catch (IOException e) {
      e.printStackTrace();
    }

  }

  public void touch_send(byte action, int x, int y, long ts, byte pointer) {                  // Touch event send. Called only by hu_act:touch_send()
    touch_send(action,ts,new CoordinatesHelpers(x,y,pointer, 0));
  }
 
```

And the protobuf for that is:

```
syntax="proto2";

package HU;

option java_package = "gb.xxy.hr";

option java_outer_classname = "Input";

enum MsgType
{
    MSG_INPUT_EVENT = 0x8001;
    MSG_INPUT_BINDINGREQUEST = 0x8002;
    MSG_INPUT_BINDINGRESPONSE = 0x8003;
}
enum MessageStatus
{
    STATUS_OK = 0;
}
message Key
{
    required uint32 keycode = 1;
    required bool down = 2;
    required uint32 metastate = 3;
    required bool longpress = 4;
}

message KeyEvent
{
    repeated Key keys = 1;
}

message TouchEvent
{
    enum PointerAction
    {
        TOUCH_ACTION_DOWN = 0;
        TOUCH_ACTION_UP = 1;
        TOUCH_ACTION_MOVE = 2;
        TOUCH_ACTION_CANCEL = 3;
        TOUCH_ACTION_OUTSIDE  = 4;
        TOUCH_ACTION_POINTER_DOWN  = 5;
        TOUCH_ACTION_POINTER_UP   = 6;
        // 0x6
    }

    message Pointer
    {
        optional uint32 x = 1;
        optional uint32 y = 2;
        optional uint32 pointer_id = 3;
    }
    repeated Pointer pointer_data = 1;
    optional uint32 action_index = 2;
    optional PointerAction action = 3;
}

message RelativeEvent_Rel
{
    required uint32 keycode = 1;
    required int32 delta = 2;
}

message RelativeEvent
{
    repeated RelativeEvent_Rel data = 1;
}

message AbsoluteEvent_Abs
{
    required uint32 keycode = 1;
    required int32 value = 2;
}

message AbsoluteEvent
{
    repeated AbsoluteEvent_Abs data = 1;
}

message InputReport
{
    required MsgType message_type = 1;
    required int32 disp_channel_id = 2;
    required uint64 timestamp = 3;		
    optional TouchEvent touch_event = 4;
    optional KeyEvent key_event = 5;
    optional AbsoluteEvent absolute_event = 6;
    optional RelativeEvent relative_event = 7;
    optional TouchEvent touchpad_event = 8;
}

message KeyBindingRequest
{
    repeated int32 keycodes = 1;
}
```