f1xpl/openauto #39: [Ubuntu] White screen on device connect

hitech95, 2018-03-11:
Hi,
i'm new with the open auto project. I tried to build it on my Ubuntu machine and compilation works fine.
When I try to run I got the warning screen and all looks working fine.

The problem come when I try to connect my phone. What I get is a white screen.

I'm not sure if it is a proble of aasdk or if is a problem with openauto.
I have removed the rpi flag from the cmake command because this is a Intel machine without openmax library.
This is the log:
```
nicolo@nicolo-Peppy:/opt/openauto/bin$ ./autoapp 
[2018-03-11 20:39:41.442058] [0x00007fc516570940] [warning] [AaApp] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file
[2018-03-11 20:39:41.676827] [0x00007fc4fdeed700] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-11 20:39:56.680655] [0x00007fc4fdeed700] [info]    [AaApp] [USBApp] Device connected.
[2018-03-11 20:39:56.684567] [0x00007fc4fdeed700] [info]    [AaApp] [AndroidAutoEntity] start.
[2018-03-11 20:39:56.685280] [0x00007fc516570940] [debug]   [AaApp] [AudioInput] create.
[2018-03-11 20:39:56.707309] [0x00007fc516570940] [debug]   [AaApp] [AudioOutput] create.
[2018-03-11 20:39:56.716721] [0x00007fc516570940] [debug]   [AaApp] [AudioOutput] create.
[2018-03-11 20:39:56.717226] [0x00007fc516570940] [debug]   [AaApp] [AudioOutput] create.
[2018-03-11 20:39:56.717628] [0x00007fc516570940] [debug]   [AaApp] [QtVideoOutput] create.
[2018-03-11 20:39:56.850444] [0x00007fc4fdeed700] [info]    [AaApp] [AudioInputService] start.
[2018-03-11 20:39:56.850681] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] start, channel: MEDIA_AUDIO
[2018-03-11 20:39:56.850765] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] start, channel: SPEECH_AUDIO
[2018-03-11 20:39:56.850825] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] start, channel: SYSTEM_AUDIO
[2018-03-11 20:39:56.850890] [0x00007fc4fdeed700] [info]    [AaApp] [SensorService] start.
[2018-03-11 20:39:56.850960] [0x00007fc4fdeed700] [info]    [AaApp] [VideoService] start.
[2018-03-11 20:39:56.851024] [0x00007fc4fdeed700] [info]    [AaApp] [BluetoothService] start.
[2018-03-11 20:39:56.851085] [0x00007fc4fdeed700] [info]    [AaApp] [InputService] start.
[2018-03-11 20:39:56.991836] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-03-11 20:39:56.991965] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] Begin handshake.
[2018-03-11 20:39:57.010832] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 2311
[2018-03-11 20:39:57.015474] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] continue handshake.
[2018-03-11 20:39:57.036469] [0x00007fc4f7fff700] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 51
[2018-03-11 20:39:57.036617] [0x00007fc4f7fff700] [info]    [AaApp] [AndroidAutoEntity] Auth completed.
[2018-03-11 20:39:57.071567] [0x00007fc4fdeed700] [info]    [AaApp] [AndroidAutoEntity] Discovery request, device name: Android, brand: Xiaomi Redmi 4
[2018-03-11 20:39:57.071668] [0x00007fc4fdeed700] [info]    [AaApp] [AudioInputService] fill features.
[2018-03-11 20:39:57.071695] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-03-11 20:39:57.071715] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-03-11 20:39:57.071732] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-03-11 20:39:57.071790] [0x00007fc4fdeed700] [info]    [AaApp] [SensorService] fill features.
[2018-03-11 20:39:57.071811] [0x00007fc4fdeed700] [info]    [AaApp] [VideoService] fill features.
[2018-03-11 20:39:57.071836] [0x00007fc4fdeed700] [info]    [AaApp] [BluetoothService] fill features
[2018-03-11 20:39:57.071852] [0x00007fc4fdeed700] [info]    [AaApp] [InputService] fill features.
[2018-03-11 20:39:57.334714] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] requested audio focus, type: 4
[2018-03-11 20:39:57.334769] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] audio focus state: 3
[2018-03-11 20:39:57.357677] [0x00007fc4fdeed700] [info]    [AaApp] [AudioInputService] open request, priority: 0
[2018-03-11 20:39:57.357709] [0x00007fc4fdeed700] [info]    [AaApp] [AudioInputService] open status: 0
[2018-03-11 20:39:57.357977] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-03-11 20:39:57.358000] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2018-03-11 20:39:57.358466] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2018-03-11 20:39:57.358560] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] open status: 0, channel: SPEECH_AUDIO
[2018-03-11 20:39:57.358833] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2018-03-11 20:39:57.358924] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[2018-03-11 20:39:57.359401] [0x00007fc4fceeb700] [info]    [AaApp] [SensorService] open request, priority: 0
[2018-03-11 20:39:57.359467] [0x00007fc4fceeb700] [info]    [AaApp] [SensorService] open status: 0
[2018-03-11 20:39:57.360070] [0x00007fc4fceeb700] [info]    [AaApp] [VideoService] open request, priority: 0
[2018-03-11 20:39:57.367233] [0x00007fc4fceeb700] [info]    [AaApp] [VideoService] open status: 0
[2018-03-11 20:39:57.367779] [0x00007fc4fdeed700] [info]    [AaApp] [InputService] open request, priority: 0
[2018-03-11 20:39:57.367802] [0x00007fc4fdeed700] [info]    [AaApp] [InputService] open status: 0
[2018-03-11 20:39:57.368092] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] setup request, channel: MEDIA_AUDIO, config index: 1
[2018-03-11 20:39:57.368114] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] setup status: 2, channel: MEDIA_AUDIO
[2018-03-11 20:39:57.368356] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] setup request, channel: SPEECH_AUDIO, config index: 1
[2018-03-11 20:39:57.368390] [0x00007fc4fdeed700] [info]    [AaApp] [AudioService] setup status: 2, channel: SPEECH_AUDIO
[2018-03-11 20:39:57.368577] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] setup request, channel: SYSTEM_AUDIO, config index: 1
[2018-03-11 20:39:57.368593] [0x00007fc4fceeb700] [info]    [AaApp] [AudioService] setup status: 2, channel: SYSTEM_AUDIO
[2018-03-11 20:39:57.369680] [0x00007fc4fceeb700] [info]    [AaApp] [VideoService] setup request, config index: 3
[2018-03-11 20:39:57.369718] [0x00007fc4fceeb700] [info]    [AaApp] [VideoService] setup status: 2
[2018-03-11 20:39:57.369906] [0x00007fc4fceeb700] [info]    [AaApp] [VideoService] video focus indication.
[2018-03-11 20:39:57.370645] [0x00007fc4f7fff700] [info]    [AaApp] [SensorService] sensor start request, type: 13
[2018-03-11 20:39:57.370880] [0x00007fc4f7fff700] [info]    [AaApp] [InputService] binding request, scan codes count: 0
[... 76 lines cut ...]
appsrc: push buffer wrong state
appsrc: push buffer wrong state
[2018-03-11 20:40:04.606696] [0x00007fc4fd6ec700] [error]   [AaApp] [VideoService] channel error: AaSdk error code: 10, native code: 1
[2018-03-11 20:40:04.606836] [0x00007fc4fd6ec700] [error]   [AaApp] [AndroidAutoEntity] channel error: AaSdk error code: 10, native code: 1
[2018-03-11 20:40:04.606895] [0x00007fc4fd6ec700] [info]    [AaApp] [USBApp] quit.
[2018-03-11 20:40:04.606931] [0x00007fc4fd6ec700] [info]    [AaApp] [AndroidAutoEntity] stop.
[2018-03-11 20:40:04.606989] [0x00007fc4fd6ec700] [info]    [AaApp] [VideoService] stop.
[2018-03-11 20:40:04.610520] [0x00007fc4f7fff700] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 10, native code: 1
[2018-03-11 20:40:04.610628] [0x00007fc4f7fff700] [error]   [AaApp] [SensorService] channel error: AaSdk error code: 10, native code: 1
[2018-03-11 20:40:04.610687] [0x00007fc4f7fff700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SYSTEM_AUDIO
[2018-03-11 20:40:04.610733] [0x00007fc4f7fff700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SPEECH_AUDIO
[2018-03-11 20:40:04.610768] [0x00007fc4f7fff700] [error]   [AaApp] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: MEDIA_AUDIO
[... cut ...]
[2018-03-11 20:40:04.611095] [0x00007fc4f7fff700] [info]    [AaApp] [AudioService] stop, channel: SPEECH_AUDIO
[2018-03-11 20:40:04.611142] [0x00007fc4f7fff700] [info]    [AaApp] [AudioService] stop, channel: MEDIA_AUDIO
[2018-03-11 20:40:04.611189] [0x00007fc4f7fff700] [info]    [AaApp] [AudioInputService] stop.
[2018-03-11 20:40:04.611237] [0x00007fc4f7fff700] [info]    [AaApp] [BluetoothService] stop.
[2018-03-11 20:40:04.611649] [0x00007fc4fd6ec700] [info]    [AaApp] [USBApp] Waiting for device...
[2018-03-11 20:40:04.616108] [0x00007fc4fd6ec700] [debug]   [AaApp] [AndroidAutoEntity] destroy.
```

f1xpl, 2018-03-11:
Hello,
Did you install all needed Qt media libraries (apt-get install libqt5multimedia5 libqt5multimedia5-plugins libqt5multimediawidgets5 qtmultimedia5-dev)?

Please check whether H.264 video codec is installed on your system. I am testing OpenAuto with Ubuntu 17.10 installed on VMWare Player and it works fine. 

Also you can try to install libav for gstreamer (sudo apt-get install gstreamer1.0-libav).

hitech95, 2018-03-11:
Yes, I have all the libs installed:
```
nicolo@nicolo-Peppy:~$ sudo apt-get install libqt5multimedia5 libqt5multimedia5-plugins libqt5multimediawidgets5 qtmultimedia5-dev gstreamer1.0-libav
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
libqt5multimedia5 is already the newest version (5.9.1-3ubuntu1).
libqt5multimedia5-plugins is already the newest version (5.9.1-3ubuntu1).
libqt5multimediawidgets5 is already the newest version (5.9.1-3ubuntu1).
qtmultimedia5-dev is already the newest version (5.9.1-3ubuntu1).
gstreamer1.0-libav is already the newest version (1.12.2-1).
0 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
nicolo@nicolo-Peppy:~$
```

f1xpl, 2018-03-11:
As far as I remember h264 codec could be also included in "gstreamer1.0-plugins-bad" package. Please install it (sudo apt-get install gstreamer1.0-plugins-bad) and test.

hitech95, 2018-03-11:
Already installed:
nicolo@nicolo-Peppy:/opt/openauto_build$ sudo apt-get install gstreamer1.0-plugins-bad
[sudo] password di nicolo: 
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
gstreamer1.0-plugins-bad is already the newest version (1.12.3-1ubuntu1).
0 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
nicolo@nicolo-Peppy:/opt/openauto_build$

f1xpl, 2018-03-11:
Take a look on this issue https://bugs.launchpad.net/ubuntu/+source/gstreamer-vaapi/+bug/1720820

It could be related to your graphic card driver and hardware acceleration of video playback.

hitech95, 2018-03-11:
Yup, you are right it was a problem with the graphic driver I have resolved removing the gstreamer acceleration pkg

closed 2018-03-11 by hitech95

hitech95, 2018-03-14:
Just a quick update in this, the video driver and hw acceleration for h264 with vaapi is working fine on my setup.
Also Gstreamer is working with vaapi. Looks like that something is broken in qt. 
I'm not an expert so if you want we can try to debug it.

 Unfortunately my cpu is not fast enough to keep a smooth playback.