f1xpl/openauto #42: Support of air mouse, custom key binding

Dennisd80, 2018-03-18:
Would it be possible to add the feature to allow custom key bindings, because i have a vehicle without touchscreen but regular screen, and would like to have the AA on that screen. I have tried it on my normal television to test it with an airmouse and regular usb mouse but unfortunally without the result i'm looking for. The airmouse after changing the key bindings for up,down,left and right to true did not function like i wanted (up is google assistant etc). With the regular usb mouse i was unable to see the cursor (even after enable it and following the steps in the wiki).

f1xpl, 2018-03-18:
If you are able to bind desired keys to your device then I think it is not needed to do it in OpenAuto. Unless you are able to bind only particular key codes then it would be needed.

The most important keyboard keys needed to use Android Auto without touchscreen are:

1 - Simulate left turn of rotary controller
2- Simulate right tun of rotary controller

You need above keys to manage focus of buttons and menu options visible on Android Auto interface

Left arrow - Open and close context menu
Down/Up arrows - Switch "segments" (menu bar or currently running application) of Android auto interface

I suggest you to connect keyboard and get some practice in controlling Android Auto in a way different than touchscreen.

Finally, if you are able to bind mentioned keys to your device then it is not needed to add such functionality to OpenAuto.

Dennisd80, 2018-03-18:
Thx, you're absolutely right. Thx for the quick response

closed 2018-03-19 by f1xpl