f1xpl/openauto #43: aasdk_build Make error

x3graphics, 2018-03-23:
Getting this error on the make command for aasdk

pi@AUTOPI:~/aasdk_build $ make
[  0%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/VideoResolutionEnum.proto
VideoResolutionEnum.proto:19:8: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
aasdk_proto/CMakeFiles/aasdk_proto.dir/build.make:822: recipe for target 'aasdk_proto/VideoResolutionEnum.pb.cc' failed
make[2]: *** [aasdk_proto/VideoResolutionEnum.pb.cc] Error 1
CMakeFiles/Makefile2:122: recipe for target 'aasdk_proto/CMakeFiles/aasdk_proto.dir/all' failed
make[1]: *** [aasdk_proto/CMakeFiles/aasdk_proto.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

f1xpl, 2018-03-24:
Please use newer version of protobuf compiler.

closed 2018-03-24 by f1xpl

x3graphics, 2018-03-27:
I was installing on jessie, my bad