f1xpl/openauto #44: Bottom right button should minimize or quit app

ryanlovett, 2018-03-23:
First, thanks very much for this awesome app. I just got it working on an rpi2 last night.

When I press Android Auto's bottom right button on the head unit in my car, a button appears that says something like "Return to Pioneer". This makes AA go away when I press it so that I can operate the other features of the head unit.

When I press the same button in openauto, a button appears that says something like "Return to OpenAuto". When I press that, openauto does not quit. It'd be nice if the app could either minimize or quit itself when that button is pushed.

Maju3, 2018-03-24:
Yes i think in the future there will be plugins for open auto.like a basic music player to play files from the PIs sd card.or obd apps you know.so i think in the future the button right on bottom will close android auto and you can use other apps.same like your pioneer headunit works

parkerlreed, 2018-03-26:
OpenAuto is literally just for streaming from the phone (video at a minimum, plus audio channels if needed)

Everything you see on the screen besides for the initial menu/settings pages is completely rendered on the phone. Anything you want to add there would have to be able to be done within Android Auto itself. 

There are existing projects if you want more of a standalone head unit system. http://headunit.viktorgino.me/

f1xpl, 2018-03-29:
Relatedhttps://github.com/f1xpl/openauto/issues/33