f1xpl/openauto #46: x86_64 Fedora 27 build

brianjmurrell, 2018-03-24:
So, I've built on x86_64 Fedora 27.  When I run `./bin/autoapp` the UI seems to come up on my screen and I can interact with it, enabling a cursor and opening settings etc., but even when I plug my phone, even when it's running Android Auto into a USB port the UI continues to say _Waiting for device_ with:

```
[2018-03-23 23:02:12.287588] [0x00007fab24906fc0] [warning] [OpenAuto] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
[2018-03-23 23:02:12.301874] [0x00007faaee7fc700] [info]    [OpenAuto] [App] Waiting for device...
[2018-03-23 23:02:12.303365] [0x00007faaedffb700] [info]    [OpenAuto] [App] Devices enumeration result: false
```

from the app.

The phone is seen on USB:

```
Bus 003 Device 006: ID 18d1:4ee7 Google Inc. 
```

Is there something more/else I need to do?

f1xpl, 2018-03-24:
Most probably you must setup udev rules. Take a look on the wiki how to do it.

brianjmurrell, 2018-03-24:
Yeah.  I did that.  The USB device gets permissions so that my user can read it:

```shell
$ ls -l /dev/bus/usb/003/018
crw-rw----+ 1 root brian 189, 273 Mar 24 09:15 018
```

and openauto opens it and reads it:

```strace
12371 openat(AT_FDCWD, "/dev/bus/usb/003/018", O_RDWR) = 23
12371 ioctl(23, USBDEVFS_GET_CAPABILITIES, 0x7fa2580035f4) = 0
12371 ioctl(23, USBDEVFS_SUBMITURB <unfinished ...>
...
```

But still stays stuck _Waiting for device_.

Is there any debug logging I can enable to get more detail about what it's trying to do?

brianjmurrell, 2018-03-26:
Any further ideas or advise here?

closed 2018-04-05 by f1xpl

brianjmurrell, 2018-04-06:
You closed this without even trying to provide any further advise at getting it working?

f1xpl, 2018-04-06:
This issue is mostly related to the udev and USB handling in the Fedora. If phone is not detected by OpenAuto it means that libusb did not catch any hotplug event. There are many reasons why hotplug was not detected and actually it is not a scope of OpenAuto at all.

brianjmurrell, 2018-04-06:
> libusb did not catch any hotplug event

But it must be working because when I plug my phone into my machine and then change the phone from _USB Charging_ to "Transfer files" Nautilus sees the phone and let's me view the files on it.