f1xpl/openauto #52: Night mode

aledthom, 2018-03-28:
Night mode on AA seems to be unreliable, not always switching to night colours when it's dark. Is it working from the phone time and location or the pi time (my pi doesn't have a rtc), or is it using the light sensor on the phone?

Some OEM head units use the cars light sensor (auto headlights sensor) to trigger night mode in AA, would it be possible to implement a night mode on/off option- perhaps with a hot key? This would enable me to wire it to work from the cars sensor or headlights.
Thanks.

f1xpl, 2018-03-28:
https://github.com/f1xpl/openauto/issues/33

f1xpl, 2018-09-11:
OpenAuto Pro (new project that I am involved in) has an option to toggle day/night mode directly from the UI. Here is an overview video https://www.youtube.com/watch?v=9sTOMI1qTiA where you can see how it works.

manvirrr, 2018-12-04:
So is the Opensource Project dead????

smartroad, 2018-12-06:
I'd ask the same is this opensource version dead? Seems like a basic feature that should happen to stop the screen being glaringly bright at night is missing. Can this fix be reversed into the open source version or is it just have to pay now? :(

manvirrr, 2018-12-06:
I found a way to attach a sensor and have openauto (open source) change from night to day or day to night while the phone is connected... 

This should be implemented in crankshaft ng soon...

https://www.reddit.com/r/crankshaft/comments/a3alak/daynight_mode_easy_switch/

manvirrr, 2018-12-06:
I just figured out a way go attach a sensor and have openauto change from day to night mode without disconnecting the phone.. this will be implemented soon in crankshaft ng...

abchet, 2019-01-10:
@smartroad i fixed mine by open AA in phone>developer setting>day/night>phone control

manvirrr, 2019-01-10:
I tried that too but it would only switch to night mode around 1 am. And back to day mode at 3 am