f1xpl/openauto #54: development branch doesn't compile

bttf92, 2018-03-28:
aasdk compiled just fine but cmake command for openauto fails with these errors

CMakeError.log:
`Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pi/openauto_build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_adb03/fast"
/usr/bin/make -f CMakeFiles/cmTC_adb03.dir/build.make CMakeFiles/cmTC_adb03.dir/build
make[1]: ingresso nella directory "/home/pi/openauto_build/CMakeFiles/CMakeTmp"
Building CXX object CMakeFiles/cmTC_adb03.dir/CheckSymbolExists.cxx.o
/usr/bin/c++     -Wall -pedantic -fPIC    -o CMakeFiles/cmTC_adb03.dir/CheckSymbolExists.cxx.o -c /home/pi/openauto_build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx
Linking CXX executable cmTC_adb03
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_adb03.dir/link.txt --verbose=1
/usr/bin/c++    -Wall -pedantic -fPIC     CMakeFiles/cmTC_adb03.dir/CheckSymbolExists.cxx.o  -o cmTC_adb03
CMakeFiles/cmTC_adb03.dir/CheckSymbolExists.cxx.o: nella funzione "main":
CheckSymbolExists.cxx:(.text+0x48): riferimento non definito a "pthread_create"
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_adb03.dir/build.make:97: set di istruzioni per l'obiettivo "cmTC_adb03" non riuscito
make[1]: *** [cmTC_adb03] Errore 1
make[1]: uscita dalla directory "/home/pi/openauto_build/CMakeFiles/CMakeTmp"
Makefile:126: set di istruzioni per l'obiettivo "cmTC_adb03/fast" non riuscito
make: *** [cmTC_adb03/fast] Errore 2

File /home/pi/openauto_build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/openauto_build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_c2651/fast"
/usr/bin/make -f CMakeFiles/cmTC_c2651.dir/build.make CMakeFiles/cmTC_c2651.dir/build
make[1]: ingresso nella directory "/home/pi/openauto_build/CMakeFiles/CMakeTmp"
Building CXX object CMakeFiles/cmTC_c2651.dir/CheckFunctionExists.cxx.o
/usr/bin/c++     -Wall -pedantic -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_c2651.dir/CheckFunctionExists.cxx.o -c /home/pi/openauto_build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx
Linking CXX executable cmTC_c2651
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c2651.dir/link.txt --verbose=1
/usr/bin/c++    -Wall -pedantic -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_c2651.dir/CheckFunctionExists.cxx.o  -o cmTC_c2651 -lpthreads
/usr/bin/ld: impossibile trovare -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_c2651.dir/build.make:97: set di istruzioni per l'obiettivo "cmTC_c2651" non riuscito
make[1]: *** [cmTC_c2651] Errore 1
make[1]: uscita dalla directory "/home/pi/openauto_build/CMakeFiles/CMakeTmp"
Makefile:126: set di istruzioni per l'obiettivo "cmTC_c2651/fast" non riuscito
make: *** [cmTC_c2651/fast] Errore 2
`

parkerlreed, 2018-03-28:
What OS/compiler versions?

bttf92, 2018-03-28:
Raspbian lite with latest updates, can't check right now the versions

parkerlreed, 2018-03-28:
Also rerunning the build with `LC_ALL=c` would help (For the output to be in English)

bttf92, 2018-03-28:
Here you go, gcc version is 6.3.0 20170516

cmake output:

-- The CXX compiler identification is GNU 6.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Found Protobuf: /usr/lib/arm-linux-gnueabihf/libprotobuf.so;-lpthread (found version "3.0.0")
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.1.0f")
CMake Error at cmake_modules/Findrtaudio.cmake:64 (message):
  Could not find rtaudio
Call Stack (most recent call first):
  CMakeLists.txt:35 (find_package)

CMakeError.log

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pi/openauto_build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_86c74/fast"
/usr/bin/make -f CMakeFiles/cmTC_86c74.dir/build.make CMakeFiles/cmTC_86c74.dir/build
make[1]: Entering directory '/home/pi/openauto_build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_86c74.dir/CheckSymbolExists.cxx.o
/usr/bin/c++     -Wall -pedantic -fPIC    -o CMakeFiles/cmTC_86c74.dir/CheckSymbolExists.cxx.o -c /home/pi/openauto_build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx
Linking CXX executable cmTC_86c74
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_86c74.dir/link.txt --verbose=1
/usr/bin/c++    -Wall -pedantic -fPIC     CMakeFiles/cmTC_86c74.dir/CheckSymbolExists.cxx.o  -o cmTC_86c74
CMakeFiles/cmTC_86c74.dir/CheckSymbolExists.cxx.o: In function `main':
CheckSymbolExists.cxx:(.text+0x48): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_86c74.dir/build.make:97: recipe for target 'cmTC_86c74' failed
make[1]: *** [cmTC_86c74] Error 1
make[1]: Leaving directory '/home/pi/openauto_build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_86c74/fast' failed
make: *** [cmTC_86c74/fast] Error 2

File /home/pi/openauto_build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/openauto_build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_c3b4e/fast"
/usr/bin/make -f CMakeFiles/cmTC_c3b4e.dir/build.make CMakeFiles/cmTC_c3b4e.dir/build
make[1]: Entering directory '/home/pi/openauto_build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_c3b4e.dir/CheckFunctionExists.cxx.o
/usr/bin/c++     -Wall -pedantic -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_c3b4e.dir/CheckFunctionExists.cxx.o -c /home/pi/openauto_build/CMakeFiles/CheckLibraryExists/CheckFunctionExists.cxx
Linking CXX executable cmTC_c3b4e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c3b4e.dir/link.txt --verbose=1
/usr/bin/c++    -Wall -pedantic -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_c3b4e.dir/CheckFunctionExists.cxx.o  -o cmTC_c3b4e -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_c3b4e.dir/build.make:97: recipe for target 'cmTC_c3b4e' failed
make[1]: *** [cmTC_c3b4e] Error 1
make[1]: Leaving directory '/home/pi/openauto_build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_c3b4e/fast' failed
make: *** [cmTC_c3b4e/fast] Error 2

parkerlreed, 2018-03-28:
See if it clears up by installing `librtaudio-dev`

CMake output shows that as missing (and it's a new requirement)

bttf92, 2018-03-28:
Can't install, it's missing from raspbian repositories

parkerlreed, 2018-03-28:
There are armhf builds... It should be available https://packages.debian.org/jessie/librtaudio-dev

bttf92, 2018-03-29:
Raspbian uses custom repositories and, as far as I know, doesn't have all the packages listed in debians ones

f1xpl, 2018-03-29:
@bttf92 It looks like something is wrong with your environment. First of all you should use Raspbian Stretch Desktop. rtaudio packages are available at least on Raspbian Stretch. Please try to build aasdk and openauto using recommended Raspbian version and in case of any problems reopen this issue.

closed 2018-03-29 by f1xpl