f1xpl/openauto #55: Autostart Openauto when Wifi connected

Vitalic66, 2018-03-30:
Is it possible to build in a feature, if wifi connection is true and connected to recent wifi then start. (Leave disclaimer).
Background: i setup a tasker task to start wifi hotspot on phone when BT is nearby. So pi connects to phone hotspot (auto). Also AA server starts with this task (rooted phone).

parkerlreed, 2018-03-30:
Maybe just have recent WiFi addresses part of the enumeration logic for USB. 

@Vitalic66 Did you ever figure out how to script stopping the server gracefully?

Vitalic66, 2018-03-30:
@parkerlreed this works for me:

am stopservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService

parkerlreed, 2018-03-30:
Problem I've run into is, using stop the only way you can restart the server is force stopping the application. 

````
angler:/ # am startservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService
Starting service: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] cmp=com.google.android.projection.gearhead/.companion.DeveloperHeadUnitNetworkService }
angler:/ # am stopservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService                                                                                          
Stopping service: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] cmp=com.google.android.projection.gearhead/.companion.DeveloperHeadUnitNetworkService }
Service stopped
255|angler:/ # am startservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService                                                                                     
Starting service: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] cmp=com.google.android.projection.gearhead/.companion.DeveloperHeadUnitNetworkService }
angler:/ # am stopservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService                                                                                          
Stopping service: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] cmp=com.google.android.projection.gearhead/.companion.DeveloperHeadUnitNetworkService }
Service not stopped: was not running.
````

You can't use stop and then start again right after.

parkerlreed, 2018-03-30:
Ok so adding in an am force-stop to my task flow does the trick

Start 
````
am startservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService
````

Stop
````
am stopservice -W com.google.android.projection.gearhead/com.google.android.projection.gearhead.companion.DeveloperHeadUnitNetworkService
am force-stop com.google.android.projection.gearhead
````

f1xpl, 2018-05-03:
@ulfdogg This issue was added to the backlog. Resolve date was not estimated yet.

ppietak, 2019-04-12:
>EDIT: Ok so that works fine via adb shell but for some reason it fails every time in Tasker when running the stop service (in that same order as above). Weird :/

I have experienced same issue.
It looks like Android Auto service isn't `exported`, so it can't be simply started from outside, if package/component isn't running at the moment.

I've found a workaround for this, also using Tasker, which works for me: 
- Before running `am startservice ...` command,
use "Launch App" with chosen `Android Auto:com.google.android.gearhead.vanagon.home1` (appears after long press on some of the AA icons).

It's a hack, cause it uses AA internal component names, so be aware that it may stop working in the future. Nevertheless, it's working on Android Auto 4.1.