f1xpl/openauto #56: Communication Error 2

danbiba, 2018-03-31:
Android Auto on my Raspberry pi is saying Waiting for Device, but the phone is saying "Android Auto encountered an error  Communication error 2 - the phone and the car are running incompatible software"  this happens on an older android LG and my newer Moto Z Play Droid. Any ideas?

majt3, 2018-04-04:
I had a similar issue today.  This turned out to be that I was trying to run the emulator at 1080p but hadn't enabled that resolution in the developer settings of the Android Auto app on the phone.

f1xpl, 2018-04-04:
What is your setup? Do you have USB sound card connected? Please provide more details.

majt3, 2018-04-05:
In my case I am using a Pi 3 with additional hardware except a keyboard, mouse and 1080p display.

closed 2018-04-05 by f1xpl