f1xpl/openauto #57: Plugging in Pixel XL does not trigger connected device

sethraymond, 2018-03-31:
I have plugged in my phone, and my Pi has recognized that a device has been plugged in, but OpenAuto is still waiting for a device. My phone says that AndroidAuto has connected to my car (the Pi running OpenAuto). Do I need to change a setting on my phone on how the USB connection works?

cwasilei, 2018-04-01:
have you done the [udev rule](https://github.com/f1xpl/openauto/wiki/udev-rules-(USB-permissions) step?

sethraymond, 2018-04-01:
Yep!
```
$ cat /etc/udev/rules.d/openauto.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="*", ATTR{idProduct}=="*", MODE="0660", GROUP="plugdev"
```

f1xpl, 2018-04-04:
What is your setup? Do you have USB sound card connected?

closed 2018-04-05 by f1xpl

sethraymond, 2018-04-06:
Can you be more specific as to what details you would like?

No, I do not have an external USB sound card connected (yet). I am planning on using one in the future, though.

Can I ask why the issue has been closed?

f1xpl, 2018-04-06:
@sethraymond, please take a look at the required hardware wiki's page. USB soundcard with microphone input is mandatory in order to use OpenAuto.

sethraymond, 2018-04-06:
Sure, I understand that I would need a USB sound card and a microphone if I wanted to use voice commands - is that really causing this issue, though? I plug my phone in and it seems like OpenAuto doesn't even recognize my device.

rafaelgil83, 2019-04-14:
@sethraymond did you ever figure this out? I'm having the same issue with the Pixel XL

sethraymond, 2019-04-15:
@rafaelgil83 I have not figured this out yet, no. I figured it was just my Pixel XL that was causing issues, but apparently it seems to be more widespread than that. Are you using a 1st gen Pixel XL?

rafaelgil83, 2019-04-15:
Yes, I have the first Gen XL and openauto running on a rasp pi, I read somewhere it could be bc the pi doesn't have a mic, that it should work with USB Sound Card? I'll have to get one and test. Did you have one plugged in?

sethraymond, 2019-04-15:
I did use a USB sound card, but didn't plug a microphone into it... Perhaps that was the issue?

rafaelgil83, 2019-04-28:
@sethraymond it is confirmed, I got a usb sound card with a mic and it is working. So it's not a phone issue, I guess at some point it fails if a mic is not detected.

sethraymond, 2019-05-12:
@rafaelgil83 I've tried it on mine with a USB sound card and a microphone attached. When I plug my phone it, it recognizes the Android Auto device and my Pi goes into the actual Android Auto screen. However, after about 5 or 6 seconds, it goes back to the landing screen. The first error I see is:

```
[0x6fafe310] [error]    [OpenAuto] [AndroidAutoEntity] ping timer exceeded.
```

I'll have to go digging into the code to see where that error could possibly be occurring...

parkerlreed, 2019-05-13:
@sethraymond you need to build with the proposed merge https://github.com/f1xpl/openauto/pull/170