f1xpl/openauto #58: openauto freezes if connected wireless and loses wifi

WarDrake, 2018-04-02:
If the wireless connection is lost (walk away and take the phone after done driving) while openauto is connected wirelessly the application freezes and all input is disregarded, be it mouse/keyboard/touch

can't connect trough ssh anymore either, but if you stay within range for a couple minutes the system connects back to the wifi, however openauto stays frozen and the system not responsive.

f1xpl, 2018-04-04:
Looks like the problem is related to the TCP keep alive mechanism.

f1xpl, 2018-04-05:
Correction has been merged to the development branch. OpenAuto will close connection after ~10 seconds when broken link has been detected. In order to test it please clone and compile latest development branch of aasdk and openauto projects.

WarDrake, 2018-04-06:
I checked out and compiled both Development versions, Fix works as intended, after a few seconds of the wifi being lost, openauto now returns to the connection screen and doesn't freeze anymore.

f1xpl, 2018-04-07:
Merged to master.

closed 2018-04-07 by f1xpl