f1xpl/openauto #60: White screen after a few seconds

cezarylaksa, 2018-04-05:
I'm running OpenAuto on Tinkerboard with TinkerOS beta 2.0.4 (Debian stretch). It starts and runs smooth in different resolution and fps, but after some time (usually 5-30 sec) whole screen goes white.
```
[2018-04-05 10:07:05.974659] [0xae90e310] [info]    [OpenAuto] [App] Waiting for device...
[2018-04-05 10:07:05.982378] [0xae10e310] [info]    [OpenAuto] [App] Devices enumeration result: true
[2018-04-05 10:07:06.486356] [0xae10e310] [info]    [OpenAuto] [App] Device connected.
[2018-04-05 10:07:06.504202] [0xae10e310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-04-05 10:07:06.514700] [0xb1b14ab0] [debug]   [OpenAuto] [AudioInput] create.
[2018-04-05 10:07:06.550121] [0xb1b14ab0] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-04-05 10:07:06.550644] [0xb1b14ab0] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-04-05 10:07:06.551055] [0xb1b14ab0] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-04-05 10:07:06.551493] [0xb1b14ab0] [debug]   [OpenAuto] [QtVideoOutput] create.
[2018-04-05 10:07:06.741653] [0xae10e310] [info]    [OpenAuto] [AudioInputService] start.
[2018-04-05 10:07:06.741937] [0xae10e310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-04-05 10:07:06.741971] [0xae10e310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-04-05 10:07:06.741995] [0xae10e310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-04-05 10:07:06.742018] [0xae10e310] [info]    [OpenAuto] [SensorService] start.
[2018-04-05 10:07:06.742043] [0xae10e310] [info]    [OpenAuto] [VideoService] start.
[2018-04-05 10:07:06.742066] [0xae10e310] [info]    [OpenAuto] [BluetoothService] start.
[2018-04-05 10:07:06.742089] [0xae10e310] [info]    [OpenAuto] [InputService] start.
[2018-04-05 10:07:07.260532] [0xae90e310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-04-05 10:07:07.260597] [0xae90e310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-04-05 10:07:07.464347] [0xae10e310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2018-04-05 10:07:07.513116] [0xae10e310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-04-05 10:07:07.646064] [0xad90e310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-04-05 10:07:07.647798] [0xad90e310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-04-05 10:07:07.863104] [0xae90e310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: Xiaomi MI 5
[2018-04-05 10:07:07.863186] [0xae90e310] [info]    [OpenAuto] [AudioInputService] fill features.
[2018-04-05 10:07:07.863311] [0xae90e310] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-04-05 10:07:07.863359] [0xae90e310] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-04-05 10:07:07.863391] [0xae90e310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-04-05 10:07:07.863420] [0xae90e310] [info]    [OpenAuto] [SensorService] fill features.
[2018-04-05 10:07:07.863466] [0xae90e310] [info]    [OpenAuto] [VideoService] fill features.
[2018-04-05 10:07:07.863505] [0xae90e310] [info]    [OpenAuto] [BluetoothService] fill features
[2018-04-05 10:07:07.863531] [0xae90e310] [info]    [OpenAuto] [InputService] fill features.
[2018-04-05 10:07:08.231389] [0xae10e310] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2018-04-05 10:07:08.231451] [0xae10e310] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2018-04-05 10:07:08.260712] [0xaf10e310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2018-04-05 10:07:08.260774] [0xaf10e310] [info]    [OpenAuto] [AudioInputService] open status: 0
[2018-04-05 10:07:08.262043] [0xad90e310] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-04-05 10:07:08.262108] [0xad90e310] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel count: 2
[2018-04-05 10:07:08.262198] [0xad90e310] [info]    [OpenAuto] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2018-04-05 10:07:08.262753] [0xae10e310] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUD
(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(autoapp:4733): GStreamer-CRITICAL **: gst_mini_object_unref: assertion 'mini_object != NULL' failed
Error: "Internal data stream error."
appsrc: push buffer wrong state
appsrc: push buffer wrong state

** (autoapp:4733): CRITICAL **: gst_x_image_sink_xwindow_clear: assertion 'xwindow != NULL' failed
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
...
//lot of repetitions
...
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
IO, priority: 0
[... 41 lines cut ...]
mpp_thread: MPP_THREAD_STOPPING status set mThread 0x7723e470
mpp_dec: mpp_dec_hal_thread exit ok
[2018-04-05 10:07:28.481020] [0xae90e310] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 10, native code: 1
[2018-04-05 10:07:28.481086] [0xaf10e310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 10, native code: 1
[2018-04-05 10:07:28.481140] [0xae10e310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SYSTEM_AUDIO
[2018-04-05 10:07:28.481343] [0xae90e310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SPEECH_AUDIO
[2018-04-05 10:07:28.481393] [0xad90e310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 10, native code: 1
[2018-04-05 10:07:28.481422] [0xae90e310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 10, native code: 1
[2018-04-05 10:07:28.481470] [0xae90e310] [info]    [OpenAuto] [App] quit.
[2018-04-05 10:07:28.481502] [0xae10e310] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 10, native code: 1
[2018-04-05 10:07:28.481527] [0xae90e310] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2018-04-05 10:07:28.481568] [0xad90e310] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 10, native code: 1
[2018-04-05 10:07:28.481638] [0xaf10e310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: MEDIA_AUDIO
[2018-04-05 10:07:28.481701] [0xae90e310] [info]    [OpenAuto] [AudioInputService] stop.
[2018-04-05 10:07:28.481804] [0xae90e310] [info]    [OpenAuto] [AudioService] stop, channel: MEDIA_AUDIO
[2018-04-05 10:07:28.481871] [0xae90e310] [info]    [OpenAuto] [AudioService] stop, channel: SPEECH_AUDIO
[2018-04-05 10:07:28.481931] [0xae90e310] [info]    [OpenAuto] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-04-05 10:07:28.481991] [0xae90e310] [info]    [OpenAuto] [SensorService] stop.
[2018-04-05 10:07:28.482034] [0xae90e310] [info]    [OpenAuto] [VideoService] stop.
[2018-04-05 10:07:28.482101] [0xae90e310] [info]    [OpenAuto] [BluetoothService] stop.
[2018-04-05 10:07:28.482144] [0xae90e310] [info]    [OpenAuto] [InputService] stop.
[2018-04-05 10:07:28.482180] [0xae90e310] [info]    [OpenAuto] [InputDevice] stop.
[2018-04-05 10:07:28.483526] [0xae90e310] [info]    [OpenAuto] [App] Waiting for device...
[2018-04-05 10:07:28.483594] [0xae90e310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.
```

f1xpl, 2018-04-05:
Hello, is this permanent or occasional issue?

cezarylaksa, 2018-04-05:
It's permanent. Screen always goes white (even without any user input). I can still control OpenAuto blindly clicking/touching the screen. Tried AndroidAuto 3.0 and 3.1, OpenAuto/aasdk on master and development branch, different phones etc.

f1xpl, 2018-04-06:
I think this issue is related to your GStreamer instance or hardware decoder. Did you try run OpenAuto with any stable version of TinkerOS?

closed 2018-04-08 by f1xpl