f1xpl/openauto #62: mouse pointer inaccurate

brianjmurrell, 2018-04-06:
On my Fedora x86_64 build the mouse pointer doesn't seem to reflect where the mouse click is activated/read from.

I am running at 480 mode and the openauto display is full-screen on a 1600x1200 monitor.

Any ideas what to do to make the pointer and the input read more accurate?

brianjmurrell, 2018-04-06:
It seems the further away from the top of the display I get the greater the discrepancy is.  If I click near the top of the display, the click is read where the pointer is but if I click at the bottom it gets read as about 2/3 down the screen.

brianjmurrell, 2018-04-06:
Hrm.  Looks like the horizontal sync between click and input is off also.

WarDrake, 2018-04-06:
It's probably a scaling issue, have you tried reducing the monitors resolution to 800x600 and checking if the same thing happens or if it gets better?

f1xpl, 2018-04-20:
Issue is not reproducible. Please provide more details.

brianjmurrell, 2018-04-20:
What resolution of screen are you unable to reproduce on?  I'm running on a screen which is 1600x1200.

f1xpl, 2018-07-10:
Problem is not reproducible with 1600x1200 resolution. In case of new findings, please reopen this issue.

closed 2018-07-10 by f1xpl