f1xpl/openauto #64: GPS share from RPI to Phone

rubenverhoef, 2018-04-09:
Just a question and maybe an feature request.
Is it possible to share the GPS from my Raspberry pi to my phone?
I'm using an ublox neo 8 with external roof antenna so it's really aqqurate.

If It is not possible right now, I would like to see this in next versions! Thanks.

parkerlreed, 2018-04-09:
https://github.com/f1xpl/openauto/issues/33

It would all depend on how your GPS is handled/if Android Auto itself supports that feature.

rubenverhoef, 2018-04-09:
I'm now using [HeadUnit Reloaded](https://forum.xda-developers.com/general/paid-software/android-4-1-headunit-reloaded-android-t3432348) and this software supports it. I would like to see this in openauto.
For GPS we can use the GPSD software for raspberry pi. Maybe openauto can use this data as GPS data and send it to the phone.

closed 2018-04-20 by f1xpl

rubenverhoef, 2018-04-20:
Closing the issue/feature request means this will not be implemented or just not right now?

bnevets27, 2018-09-07:
I would also like to know what closing this issue means.