f1xpl/openauto #65: Mouse pointer not showing

juliansteenbakker, 2018-04-09:
So i have installed everything on my Raspberry Pi Zero W, and everything seems to work just fine, except, no matter what i do it wont show the mouse pointer (cursor) on the screen. 

I have tried changing the omx layer to 0, as described in the wiki (https://github.com/f1xpl/openauto/wiki/Enable-visibility-of-mouse-pointer-for-Raspberry-PI-3), but this doesnt seem to work on the Raspberry Pi Zero W.

It does work fine on the "Awaiting connection" screen, but when i connect a phone it just dissapears. It does work though, but the cursor is just not showing.

f1xpl, 2018-04-20:
What is your setup? Are you using X11 or EGL?

closed 2018-05-22 by f1xpl

bklepic, 2018-06-21:
Same problem here. I use RPI 3B+ with newest jessie image

rizlas, 2019-02-20:
Same problem on rpi3b+. Just need to run some test without a touchscreen.