f1xpl/openauto #67: Is it possible to support 1024 x 600 (600p? :) )

fonsmol, 2018-04-13:
I have noticed online that a lot of people have a 1024 x 600 resolution screen, and so do I. Since I have not find a real solution for this, is it possible to support this as well? :) Im happy to try some modifications in the code my self if that helps?

WarDrake, 2018-04-13:
I have a 1024x600 screen everything works perfectly, the only screen with an issue is the one where it's waiting for a phone and only because there is wasted space the right and bottom sides, you only need to setup your screen dimensions on the config.txt file for the Pi

for reference mine are these : 

hdmi_group=2
hdmi_mode=87
hdmi_cvt=1024 600 60 6 0 0 0

That'll tell the Pi and Raspbian to use custom hdmi output screen mode with 1024 width, 600 height, 60 framerate, 15:9 aspect, no margins, progressive mode

depending on your screen you might have to mess around with those values a bit if your experience is not perfect.

fonsmol, 2018-04-13:
Hi WarDrake, I already have these settings running. Openauto runs in full screen and it does have some wasted space in the waiting screen indeed, but that is not my concern. I noticed that when running in 480p the views are a bit blurry because they seem stretched out, which sounds logical. When selecting 720p in settings the app does not work at all :(

WarDrake, 2018-04-13:
Have you enabled 720p mode on the developer settings AndroidAuto on the phone as well as on the openauto settings on the rpi?

fonsmol, 2018-04-13:
Thnx that is the best solution I guess :)

closed 2018-04-13 by fonsmol

dpnkrg, 2019-06-04:
It is possible to switch to HDMI display to GPIO Based touch screen display.

Right now, I am able to run the OpenAuto app on 3.2 Inches touch screen with Raspberry Pi3
In this, I am able to get the App Home screen, but when the AA is launched, it launched in HDMI Screen only. In short whole, Raspbian OS will display in 3.2 Inches touchscreen but the Output of AA displayed in HDMI only.

Is there any way to change the AA out to GPIO based in screen only. 

Please help !!