f1xpl/openauto #68: Voice commands are not recognized

WarDrake, 2018-04-14:
The "Ok Google" hotword works perfectly, but using USB microphone or the phone's internal Microphone no command other than the hotword is recognized, the assistant doesn't acknowledge having heard anything after the beep.

unplugging the phone and speaking to it, the voice commands immediately start working correctly so it's not the phone version.

f1xpl, 2018-04-14:
Try to use a recorder app to check whether microphone works at least at system level.

DGxInfinitY, 2018-04-14:
Same issue here. It all works until after the hot word is recognized. Then it stops listening. My mics work perfectly.

f1xpl, 2018-04-14:
How do you know that mic works at openauto level? Do you see voice indicator (this one that appears after Ok, google) waving?

WarDrake, 2018-04-14:
Yes, and if you start speaking, it'll stay open for as long as you speak, once you are done it closes, and does nothing, no matter what you said to it.

f1xpl, 2018-04-14:
So if voice from the mic is catched by the phone, openauto works as expected. It looks like problem is inside your phone. Do you have internet access when AA is running?

Please note if you connect to the WiFi without internet access, data will not be sent through 3G/LTE.

WarDrake, 2018-04-14:
No, the Internet is not a problem, since play music is running and the phone has internet, the problem seems to be OpenAuto not forwarding the voice command, since the moment you unplug the phone from open auto and talk to it, it'll work as expected, the only time it doesn't work is when the phone is connected to OpenAuto, the "Ok Google" hotword is hardcoded into the system, which is why it is recognizing it, but the rest needs to be forwarded, wich seems to be the problem.

f1xpl, 2018-04-14:
OpenAuto forwards only data grabbed from the microphone. It does not interpret it.

WarDrake, 2018-04-14:
Exactly my point, it's not forwarding the data grabbed from the mic, since the phone does work as soon as it is disconnected but not while connected to OpenAuto.

f1xpl, 2018-04-14:
Do you see voice command indicator waving when you talk to the microphone?

WarDrake, 2018-04-14:
No, when you say OK Google it opens the voice prompt, but the 4 dots are static, then they collapse and the prompt closes and it does nothing.

f1xpl, 2018-04-14:
Did you set your usb mic as a default input device in pulseaudio settings?

WarDrake, 2018-04-14:
Yeah, it is the default device.

WarDrake, 2018-04-15:
So, while copying the config file I noticed my default device was not set in there, turns out the changes on the GUI to make a device the default weren't sticking, once I made the mic the default on the config file and restarted pulseaudio it worked as expected.

since @DGxInfinitY seeems to have the same issue it might be that several people do, do you mind if I write some instructions for you to add to the sound config for the wiki ?

brett1996, 2018-04-17:
@WarDrake  Can you send me some information. i'm a real noob at these things and doesn't understand how to change such things. I have the same problem as you had.

WarDrake, 2018-04-17:
@brett1996 Here are instructions on how to get it done.

This command will print the names of the devices connected to your board
	pacmd list-sources | grep name:

Ex : name: <alsa_input.usb-CD04_CD04-01.analog-mono>

edit the file /etc/pulse/default.pa
replace the word "input" on the line that reads :
	#set-default-source input
with the name of the device without the < > brackets

Ex : set-default-source alsa_input.usb-CD04_CD04-01.analog-mono

then restart your pulseaudio instance, can be done with :
	pulseaudio -k

and restart OpenAuto. you might need to try multiple devices if not sure which one is your mic.

closed 2018-04-20 by f1xpl

azntklc, 2018-07-12:
So I´m experiencing the same issue as mentioned above and followed WarDrake´s instructions 

pi@raspberrypi:~ $ pacmd list-sources | grep name
	name: <alsa_output.usb-C-Media_Electronics_Inc._USB_Audio_Device-00.analog-stereo.monitor>
		alsa.card_name = "USB Audio Device"
		alsa.long_card_name = "C-Media Electronics Inc. USB Audio Device at usb-3f980000.usb-1.2, full speed"
		alsa.driver_name = "snd_usb_audio"
		device.vendor.name = "C-Media Electronics, Inc."
		device.product.name = "USB Audio Device"
		device.icon_name = "audio-card-usb"
	name: <alsa_input.usb-C-Media_Electronics_Inc._USB_Audio_Device-00.analog-mono>
		alsa.name = "USB Audio"
		alsa.subdevice_name = "subdevice #0"
		alsa.card_name = "USB Audio Device"
		alsa.long_card_name = "C-Media Electronics Inc. USB Audio Device at usb-3f980000.usb-1.2, full speed"
		alsa.driver_name = "snd_usb_audio"
		device.vendor.name = "C-Media Electronics, Inc."
		device.product.name = "USB Audio Device"
		device.profile.name = "analog-mono"
		alsa.mixer_name = "USB Mixer"
		device.icon_name = "audio-card-usb"
				device.icon_name = "audio-input-microphone"

used command pi@raspberrypi:~ $ sudo nano /etc/pulse/default.pa

scrolled all the way to the bottom and edited to: 

### Make some devices default
#set-default-sink output
set-default-source alsa_input.usb-C-Media_Electronics_Inc._USB_Audio_Device-00.analog-mono

saved to default.pa

restarted pulseaudio with : pulseaudio -k

Tested feature, still didn´t work. Rebooted pi with sudo reboot

Tested feature, still didn´t work so I ran a log file. 

[2018-07-11 21:24:01.305066] [0x7244c000] [warning] [OpenAuto] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
[2018-07-11 21:24:01.394672] [0x7244c000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-07-11 21:24:01.424923] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-07-11 21:24:01.431011] [0x6f2ff310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-07-11 21:24:12.423421] [0x6eaff310] [info]    [OpenAuto] [App] Device connected.
[2018-07-11 21:24:12.446269] [0x7244c000] [debug]   [OpenAuto] [AudioInput] create.
[2018-07-11 21:24:12.464244] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-07-11 21:24:12.464319] [0x6eaff310] [info]    [OpenAuto] [AudioInputService] start.
[2018-07-11 21:24:12.464847] [0x6eaff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-07-11 21:24:12.464917] [0x6eaff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-07-11 21:24:12.464975] [0x6eaff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-07-11 21:24:12.465032] [0x6eaff310] [info]    [OpenAuto] [SensorService] start.
[2018-07-11 21:24:12.465095] [0x6eaff310] [info]    [OpenAuto] [VideoService] start.
[2018-07-11 21:24:12.465159] [0x6eaff310] [info]    [OpenAuto] [BluetoothService] start.
[2018-07-11 21:24:12.465218] [0x6eaff310] [info]    [OpenAuto] [InputService] start.
[2018-07-11 21:24:12.492216] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-07-11 21:24:12.492278] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-07-11 21:24:12.504163] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2290
[2018-07-11 21:24:12.584624] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-07-11 21:24:12.590795] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-07-11 21:24:12.591553] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-07-11 21:24:12.604201] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: samsung SM-N950W
[2018-07-11 21:24:12.604305] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] fill features.
[2018-07-11 21:24:12.604433] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-07-11 21:24:12.604495] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-07-11 21:24:12.604538] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-07-11 21:24:12.604578] [0x6e2ff310] [info]    [OpenAuto] [SensorService] fill features.
[2018-07-11 21:24:12.604646] [0x6e2ff310] [info]    [OpenAuto] [VideoService] fill features.
[2018-07-11 21:24:12.604701] [0x6e2ff310] [info]    [OpenAuto] [BluetoothService] fill features
[2018-07-11 21:24:12.604741] [0x6e2ff310] [info]    [OpenAuto] [InputService] fill features.
[2018-07-11 21:24:12.660897] [0x6faff310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[... 47 lines cut ...]
[2018-07-11 21:24:40.770653] [0x6e2ff310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 10, native code: 1
[2018-07-11 21:24:40.770731] [0x6faff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: SPEECH_AUDIO
[2018-07-11 21:24:40.770791] [0x6e2ff310] [info]    [OpenAuto] [App] quit.
[2018-07-11 21:24:40.770813] [0x6faff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 10, native code: 1, channel: MEDIA_AUDIO
[... cut ...]
[2018-07-11 21:24:41.103143] [0x6e2ff310] [info]    [OpenAuto] [BluetoothService] stop.
[2018-07-11 21:24:41.103226] [0x6e2ff310] [info]    [OpenAuto] [InputService] stop.
[2018-07-11 21:24:41.103265] [0x6e2ff310] [info]    [OpenAuto] [InputDevice] stop.
[2018-07-11 21:24:41.104798] [0x6e2ff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-07-11 21:24:41.104925] [0x6e2ff310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.

Sorry for the long post, but any ideas as to why itś not working?? I know the mic works because the hotword works flawlessly everytime.

matoneill, 2018-07-24:
Ok, so I'm having the same issue. I have tried changing the default to every audio device on the list with no luck. I have also found out that the trigger word is recognised by the phone's internal microphone and not the pi's. Because the pi is always listening for it, it will always react, but when it comes time to record the audio to send off for voice recognition, it relies on the audio device being used with open auto, which doesn't seem to be working for a lot of people.

Does anyone have any life changing revelations for us? I know for a fact that Everlanders youtube channel got it working with a usb sound card and lapel mic...

computeridea, 2018-11-22:
for me everythings work now, i just followed as sayd, at first it doesn't work because i didn't write the full name of the device.. it's really long, 
and at second because i edited on GUI,  and copied the file in the folder, and i didn't noticed that it wa<sn't copy successfull because of permission issues, 
now it work really fine.. thanks everybody