f1xpl/openauto #69: Wireless Mode

hl68fx, 2018-04-15:
Hi everyone,

is it possible to use the Android Auto wireless mode?
https://www.androidpolice.com/2018/04/13/android-auto-wireless-now-enabled-nexus-pixel-devices/

Kind regards,
Tom

WarDrake, 2018-04-16:
Yes, you can use it wirelessly, you can see this on the feature list on the readme file

You'll need to enable developer settings :
Open Android Auto on the phone
Click Menu > About
click on "About Android Auto" on the header continiously until it say you've enabled developer settings
on the About screen, Click the 3 dots on the top right and select "Start head unit server" you might have to restart for this menu to be available.

Start the Hotspot functionality on your phone, connect the Pi to the phone's wireless, run openauto, and click "connect wireless" type the address of the phone (you'll need a phone the first time you do this) and from then on, when you click it, you can open it from a list.

parkerlreed, 2018-04-16:
I think the question was regarding the now official wireless support (Nexus/Pixel for the time being). There's a way to have that automated and not using the headunit server (@f1xpl is up to date on the applications but his official wireless option hasn't shown up yet for him to do testing)

inspector71, 2018-08-31:
I'm writing a wiki page for this based on Frank La Verne's work posted on Reddit. Isn't far away.

A lot of your questions may be addressed in that thread:

https://www.reddit.com/r/crankshaft/comments/99py4c/wireless_connection/

closed 2019-01-17 by f1xpl