f1xpl/openauto #72: Running OpenAuto without xserver

Daniele9575, 2018-04-18:
Hi,
I didn't understand if it is possible to run OpenAuto without having an xserver installed, and if it is how can i configure OpenAuto to run without x?

Thanks for the help.

f1xpl, 2018-04-18:
It is not officialy supported but should be possible.

poxin13, 2018-04-18:
Not 100% what you were asking for, still requires xorg installed, but managed to get it launched with booting direct to CLI, bypassing the desktop.

`sudo xinit /home/pi/openauto/bin/autoapp $* -- :0` I suppose `startx /home/pi/openauto/bin/autoapp` would also work.

Boot time to openauto does seem faster this method without booting the full desktop. Unfortunately plugging in the phone doesn't update the screen and causes touch to not work untill unplugged again.

xstsx, 2018-04-18:
The only way I know, it's recompile qt5 without xorg support and with eglfs support for raspberry pi.

closed 2018-04-20 by f1xpl