f1xpl/openauto #75: aasdk Build Error

thentertainer, 2018-04-21:
Hi

I cannot build the aasdk and get this error:

[ 81%] Linking CXX shared library /home/pi/aasdk/lib/libaasdk_proto.so
/usr/bin/ld: cannot open output file /home/pi/aasdk/lib/libaasdk_proto.so: No such file or directory
collect2: error: ld returned 1 exit status
aasdk_proto/CMakeFiles/aasdk_proto.dir/build.make:2277: recipe for target '/home/pi/aasdk/lib/libaasdk_proto.so' failed
make[2]: *** [/home/pi/aasdk/lib/libaasdk_proto.so] Error 1
CMakeFiles/Makefile2:122: recipe for target 'aasdk_proto/CMakeFiles/aasdk_proto.dir/all' failed
make[1]: *** [aasdk_proto/CMakeFiles/aasdk_proto.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

parkerlreed, 2018-04-22:
````
/usr/bin/ld: cannot open output file /home/pi/aasdk/lib/libaasdk_proto.so: No such file or directory
````
Did you build aasdk beforehand in that location? OpenAuto build is dependent on the paths being correct in the build command.

f1xpl, 2018-04-24:
Please follow build instructions carefully (build aasdk first).

closed 2018-04-24 by f1xpl

f1xpl, 2018-04-26:
did you build aasdk? Please post here outpu of building aasdk.

thentertainer, 2018-04-26:
The original post is the attempt of building aasdk. I am looking in the aasdk folder and the "lib" folder is not present. I have been using sudo in front of everything to make sure I don't miss anything

thentertainer, 2018-04-26:
I am currently starting from step 1 again. Will post output again soon.

thentertainer, 2018-04-26:
Here is the entire output from the make command

pi@raspberrypi:~/aasdk_build $ sudo make
[  1%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/VideoResolutionEnum.proto
[  1%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelData.proto
[  2%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelMessageIdsEnum.proto
[  2%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelSetupRequestMessage.proto
[  2%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelSetupResponseMessage.proto
[  3%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelSetupStatusEnum.proto
[  3%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelStartIndicationMessage.proto
[  4%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVChannelStopIndicationMessage.proto
[  4%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVInputChannelData.proto
[  5%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVInputOpenRequestMessage.proto
[  5%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVInputOpenResponseMessage.proto
[  5%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVMediaAckIndicationMessage.proto
[  6%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AVStreamTypeEnum.proto
[  6%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AbsoluteInputEventData.proto
[  7%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AbsoluteInputEventsData.proto
[  7%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AccelData.proto
[  7%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AudioConfigData.proto
[  8%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AudioFocusRequestMessage.proto
[  8%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AudioFocusResponseMessage.proto
[  9%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AudioFocusStateEnum.proto
[  9%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AudioFocusTypeEnum.proto
[ 10%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AudioTypeEnum.proto
[ 10%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/AuthCompleteIndicationMessage.proto
[ 10%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BindingRequestMessage.proto
[ 11%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BindingResponseMessage.proto
[ 11%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BluetoothChannelData.proto
[ 12%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BluetoothChannelMessageIdsEnum.proto
[ 12%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BluetoothPairingMethodEnum.proto
[ 12%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BluetoothPairingRequestMessage.proto
[ 13%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BluetoothPairingResponseMessage.proto
[ 13%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/BluetoothPairingStatusEnum.proto
[ 14%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ButtonCodeEnum.proto
[ 14%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ButtonEventData.proto
[ 15%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ButtonEventsData.proto
[ 15%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ChannelDescriptorData.proto
[ 15%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ChannelOpenRequestMessage.proto
[ 16%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ChannelOpenResponseMessage.proto
[ 16%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/CompassData.proto
[ 17%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/ControlMessageIdsEnum.proto
[ 17%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/DiagnosticsData.proto
[ 17%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/DoorData.proto
[ 18%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/DrivingStatusData.proto
[ 18%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/DrivingStatusEnum.proto
[ 19%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/EnvironmentData.proto
[ 19%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/FuelLevelData.proto
[ 20%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/GPSLocationData.proto
[ 20%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/GearData.proto
[ 20%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/GearEnum.proto
[ 21%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/GyroData.proto
[ 21%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/HVACData.proto
[ 22%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/HeadlightStatusEnum.proto
[ 22%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/IndicatorStatusEnum.proto
[ 23%] Running C++ protocol buffer compiler on /home/pi/aasdk/aasdk_proto/InputChannelData.proto
[... cut ...]
/home/pi/aasdk/src/Messenger/ChannelReceiveMessageQueue.cpp:58:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
CMakeFiles/aasdk.dir/build.make:257: recipe for target 'CMakeFiles/aasdk.dir/src/Messenger/ChannelReceiveMessageQueue.cpp.o' failed
make[2]: *** [CMakeFiles/aasdk.dir/src/Messenger/ChannelReceiveMessageQueue.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/aasdk.dir/all' failed
make[1]: *** [CMakeFiles/aasdk.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

thentertainer, 2018-04-26:
... which is now a new error...
In any case I cannot build aasdk

WarDrake, 2018-04-26:
@thentertainer 2 things, 1. do NOT use sudo, you're not building as root, you're building as pi user, 2 run the cmake command again, I've had this issue before, usually when I switch branches for compile, so as pi, not sudo, run cmake first and then make. it should work.

thentertainer, 2018-04-26:
This is what happens when I do

pi@raspberrypi:~/aasdk_build $ cmake -DCMAKE_BUILD_TYPE=Release ../aasdk
-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread;-lpthread (found version "3.5.1") 
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread (found version "3.5.1") 
-- Configuring done
CMake Error: Cannot open file for write: /home/pi/aasdk_build/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /home/pi/aasdk_build/CMakeCache.txt
CMake Error: : System Error: Permission denied

I am thinking I am just doing something very simply stupid

f1xpl, 2018-04-26:
It looks like you are using G++ 4.9 to build aasdk. Required version is 6.3.0 or newer.

WarDrake, 2018-04-26:
Also, your files belong to root now because you originally were using sudo to run the commands,hence the "CMake Error: : System Error: Permission denied" so give the pi user ownership of them again, for that one thing you'll need to use sudo again :P 
`sudo chown -R pi:pi /home/pi`

run cmake and make using pi user after that, and update your G++ as @f1xpl  said

thentertainer, 2018-04-26:
Thank you for the feedback. I will attempt again after making the changes

thentertainer, 2018-04-27:
Problems resolved. Thanks for the help