f1xpl/openauto #76: Waiting for Device but device recognised

sfalvey, 2018-04-23:
I have just done a fresh install of openauto. The main screen stays on waiting for device but the log is showing that the device was connected and recognised. Any idea whats going on here?

libEGL warning: DRI2: failed to authenticate
[2018-04-23 21:40:47.913161] [0x72495000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-04-23 21:40:47.956545] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-04-23 21:40:47.963184] [0x6f2ff310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-04-23 21:40:54.215683] [0x6e2ff310] [info]    [OpenAuto] [App] Device connected.
[2018-04-23 21:40:54.238969] [0x72495000] [debug]   [OpenAuto] [AudioInput] create.
[2018-04-23 21:40:54.257784] [0x72495000] [debug]   [OpenAuto] [LocalBluetoothDevice] create.
[2018-04-23 21:40:54.288757] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-04-23 21:40:54.289680] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] start.
[2018-04-23 21:40:54.290687] [0x6e2ff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-04-23 21:40:54.291224] [0x6e2ff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-04-23 21:40:54.291812] [0x6e2ff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-04-23 21:40:54.292292] [0x6e2ff310] [info]    [OpenAuto] [SensorService] start.
[2018-04-23 21:40:54.292620] [0x6e2ff310] [info]    [OpenAuto] [VideoService] start.
[2018-04-23 21:40:54.292943] [0x6e2ff310] [info]    [OpenAuto] [BluetoothService] start.
[2018-04-23 21:40:54.293274] [0x6e2ff310] [info]    [OpenAuto] [InputService] start.
[2018-04-23 21:40:54.410331] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-04-23 21:40:54.410529] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-04-23 21:40:54.427256] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2018-04-23 21:40:54.504540] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-04-23 21:40:54.527629] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-04-23 21:40:54.528752] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-04-23 21:40:54.544351] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: samsung SM-G955F
[2018-04-23 21:40:54.544594] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] fill features.

f1xpl, 2018-04-23:
Please check for power issues.

sfalvey, 2018-04-24:
The Pi itself doesn't seem to have a problem. When I plug my phone into the USB, it starts blinking the red LED. Is there a dependency for openauto related to power?

Unfortunately I dont see the enable server mode options under the developer settings on my phone to try with connecting over wireless.

f1xpl, 2018-04-24:
When USB device consumes too much power (e.g. USB C device), USB controller may disconnect the device due to safety reasons.

sfalvey, 2018-04-24:
There was nothing in the messages file or dmesg about disconnects but - good to know. I'll try and track down a powered USB hub suitable for use in a car and see if that makes a difference.

WarDrake, 2018-04-24:
@sfalvey have you configured max usb current on your Pi?
try adding this to the config.txt file : max_usb_current=1
It may solve your issue if the pi is shutting down the usb controller because of the draw.

sfalvey, 2018-04-25:
Thanks for the suggestion. I hadn't tried that but it doesn't seem to have made any difference. As soon as I plug my s8+ into the USB I get the under voltage warning in dmesg and the red led flashes. I have a desktop powered USB hub at work. I'll try that one next week and see if a powered hub can resolve the under voltage.

sfalvey, 2018-04-27:
I found and enabled the "start head unit server" and attempted to connect over wifi.  The autoapp log indicates it has connected but the screen stays on waiting for device and the phone does not indicate its in remote head unit mode, i.e. android auto ui stays available on the phone.

I have also tried with a powered hub sitting between the raspberry pi and the samsung phone so the phone is not drawing any power from the pi. The autoapp log shows it has connected, the phone shows it has connected but the UI stays stubbornly on waiting for device. 

I also forgot to mention, the UI buttons dont work once its connected and I have to kill autoapp to exit. Here is the log from the wireless connection. Followed by one over USB. An strace of the process shows its spawning about 10 process ids all of which are reading and writing data.

pi@carputer:~ $ openauto/bin/autoapp 
libEGL warning: DRI2: failed to authenticate
[2018-04-27 11:07:20.585467] [0x7249d000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-04-27 11:07:20.729529] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-04-27 11:07:20.738397] [0x6e2ff310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-04-27 11:09:17.290146] [0x6f2ff310] [info]    [OpenAuto] [App] Device connected.
[2018-04-27 11:09:17.351670] [0x7249d000] [debug]   [OpenAuto] [AudioInput] create.
[2018-04-27 11:09:18.795737] [0x7249d000] [debug]   [OpenAuto] [LocalBluetoothDevice] create.
[2018-04-27 11:09:18.862891] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-04-27 11:09:18.863100] [0x6f2ff310] [info]    [OpenAuto] [AudioInputService] start.
[2018-04-27 11:09:18.868271] [0x6f2ff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-04-27 11:09:18.868596] [0x6f2ff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-04-27 11:09:18.869014] [0x6f2ff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-04-27 11:09:18.869183] [0x6f2ff310] [info]    [OpenAuto] [SensorService] start.
[2018-04-27 11:09:18.869439] [0x6f2ff310] [info]    [OpenAuto] [VideoService] start.
[2018-04-27 11:09:18.869604] [0x6f2ff310] [info]    [OpenAuto] [BluetoothService] start.
[2018-04-27 11:09:18.869819] [0x6f2ff310] [info]    [OpenAuto] [InputService] start.
[2018-04-27 11:09:18.884021] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-04-27 11:09:18.884176] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-04-27 11:09:18.921072] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2018-04-27 11:09:19.008239] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-04-27 11:09:19.038535] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-04-27 11:09:19.039660] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-04-27 11:09:19.110519] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: samsung SM-G955F
[2018-04-27 11:09:19.111023] [0x6faff310] [info]    [OpenAuto] [AudioInputService] fill features.
[2018-04-27 11:09:19.111352] [0x6faff310] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-04-27 11:09:19.111593] [0x6faff310] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-04-27 11:09:19.111810] [0x6faff310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-04-27 11:09:19.112022] [0x6faff310] [info]    [OpenAuto] [SensorService] fill features.
[2018-04-27 11:09:19.112258] [0x6faff310] [info]    [OpenAuto] [VideoService] fill features.
[2018-04-27 11:09:19.112482] [0x6faff310] [info]    [OpenAuto] [BluetoothService] fill features
[2018-04-27 11:09:19.115417] [0x6faff310] [info]    [OpenAuto] [BluetoothService] sending local adapter adress: B8:27:EB:2F:8D:51
[2018-04-27 11:09:19.118061] [0x6faff310] [info]    [OpenAuto] [InputService] fill features.
[2018-04-27 11:09:19.350290] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2018-04-27 11:09:19.350475] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2018-04-27 11:09:19.389409] [0x6faff310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2018-04-27 11:09:19.389531] [0x6faff310] [info]    [OpenAuto] [AudioInputService] open status: 0
[2018-04-27 11:09:19.390362] [0x6f2ff310] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-04-27 11:09:19.390587] [0x6f2ff310] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel count: 2
[2018-04-27 11:09:19.391300] [0x6faff310] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2018-04-27 11:09:19.391411] [0x6faff310] [debug]   [OpenAuto] [AudioService] channel: SPEECH_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-04-27 11:09:19.392229] [0x6e2ff310] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2018-04-27 11:09:19.392466] [0x6e2ff310] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[... cut ...]
[2018-04-27 11:10:34.131226] [0x6f1ff310] [info]    [OpenAuto] [VideoService] setup request, config index: 3
[2018-04-27 11:10:34.131332] [0x6f1ff310] [info]    [OpenAuto] [OMXVideoOutput] init, state: true
[2018-04-27 11:10:34.131743] [0x6f1ff310] [info]    [OpenAuto] [VideoService] setup status: 2
[2018-04-27 11:10:34.132234] [0x6e9ff310] [info]    [OpenAuto] [VideoService] video focus indication.
[2018-04-27 11:10:34.138868] [0x6e9ff310] [info]    [OpenAuto] [VideoService] start indication, session: 0

f1xpl, 2018-04-27:
Did you set memory split to 256mb?

sfalvey, 2018-04-27:
I have the following line in config.txt, I'm assuming this is the correct way of doing that..

gpu_mem=256

f1xpl, 2018-04-27:
Are you using hdmi or touchscreen?

sfalvey, 2018-04-27:
Its on HDMI at the moment.

f1xpl, 2018-04-27:
What desktop driver are you using? Try to increment layer index in OpenAuto settings (to 2 or 3).

grcoelho, 2018-04-28:
Got the same situation here.
Followed the installation guide and got to thr waiting device screen.
When i connect the phone nothing happens. The phone is charging and android auto is running, but its not being recognized using usb. Tested with wifi and it works (a bit laggy but works).

Is there something I may try to get it recognizing the phone through usb ?

closed 2018-04-28 by f1xpl

sfalvey, 2018-05-01:
This still isn't working for me. I've decided to hold on until I can get a powered USB hub hat for the PI to be absolutely sure that its not a power issue. Unfortunately the one I ordered got lost in the post :(

I was using the default driver, but have since tried the fake KMS. the openauto settings didn't change anything either. I will retry everything when the usb hub arrives.

sfalvey, 2018-05-07:
I have a powered raspberry pi usb hub connected. I can confirm that there are no power issues listed in dmesg output. I have tried using the fake KMS driver and changing the layer in the config file first to 2 and then 3. All of these have not made any difference.

The log from openauto seems to suggest that it has connected successfully, the phone says its connected but the openauto screen is staying stubbornly on "waiting for device".

What else can I try to get this working?

sfalvey, 2018-05-09:
No idea why but a fresh install and recompile and its working. I guess some of the other software I have installed for my carputer has conflicted but at least I know its not hardware related.