f1xpl/openauto #78: Black screen after connecting phone

jackongithub, 2018-04-24:
Hi
I am trying to run openauto on ubuntu bionic beaver, with a pixel 2 phone.
When I run ./autoapp and connect my phone I get a black screen on my computer, the phone knows it is connected.

I have tried re-compiling it to no success.
Could you please tell me what I am doing wrong or what is the issue?

Thanks
-Jack

Linux mini-desktop 4.15.0-19-generic #20-Ubuntu SMP Sun Apr 22 00:03:34 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

openauto.ini
[General]
HandednessOfTrafficType=0
ShowClock=false
[Video]
FPS=1
Resolution=1
ScreenDPI=140
OMXLayerIndex=1
MarginWidth=0
MarginHeight=0
[Input]
EnableTouchscreen=false
PlayButton=false
PauseButton=false
TogglePlayButton=false
NextTrackButton=false
PreviousTrackButton=false
HomeButton=false
PhoneButton=false
CallEndButton=false
VoiceCommandButton=false
LeftButton=false
RightButton=false
UpButton=false
DownButton=false
ScrollWheelButton=false
BackButton=false
EnterButton=false
[Bluetooth]
AdapterType=0
RemoteAdapterAddress=
[Audio]
MusicAudioChannelEnabled=false
SpeechAudioChannelEnabled=false
OutputBackendType=1

Console ouput
mini@mini-desktop:~/openauto/bin$ ./autoapp 
[2018-04-25 08:22:21.446242] [0x00007f132c1c0840] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-04-25 08:22:21.491378] [0x00007f131bfe7700] [info]    [OpenAuto] [App] Waiting for device...
[2018-04-25 08:22:22.496944] [0x00007f131bfe7700] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-04-25 08:22:43.886239] [0x00007f131b7e6700] [info]    [OpenAuto] [App] Device connected.
[2018-04-25 08:22:43.892995] [0x00007f132c1c0840] [debug]   [OpenAuto] [AudioInput] create.
[2018-04-25 08:22:43.916504] [0x00007f132c1c0840] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-04-25 08:22:43.916741] [0x00007f132c1c0840] [debug]   [OpenAuto] [QtVideoOutput] create.
[2018-04-25 08:22:44.210514] [0x00007f131b7e6700] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-04-25 08:22:44.210561] [0x00007f131b7e6700] [info]    [OpenAuto] [AudioInputService] start.
[2018-04-25 08:22:44.210893] [0x00007f131b7e6700] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-04-25 08:22:44.210976] [0x00007f131b7e6700] [info]    [OpenAuto] [SensorService] start.
[2018-04-25 08:22:44.211005] [0x00007f131b7e6700] [info]    [OpenAuto] [VideoService] start.
[2018-04-25 08:22:44.211042] [0x00007f131b7e6700] [info]    [OpenAuto] [BluetoothService] start.
[2018-04-25 08:22:44.211070] [0x00007f131b7e6700] [info]    [OpenAuto] [InputService] start.
[2018-04-25 08:22:44.211653] [0x00007f131b7e6700] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-04-25 08:22:44.211684] [0x00007f131b7e6700] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-04-25 08:22:44.219605] [0x00007f131a7e4700] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2018-04-25 08:22:44.230756] [0x00007f131a7e4700] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-04-25 08:22:44.236439] [0x00007f131b7e6700] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[... 111 lines cut ...]
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
[2018-04-25 08:22:58.026742] [0x00007f131bfe7700] [info]    [OpenAuto] [AndroidAutoEntity] Shutdown request, reason: 1
[2018-04-25 08:22:58.027391] [0x00007f131a7e4700] [info]    [OpenAuto] [App] quit.
[2018-04-25 08:22:58.027500] [0x00007f131a7e4700] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2018-04-25 08:22:58.027584] [0x00007f131a7e4700] [info]    [OpenAuto] [AudioInputService] stop.
[2018-04-25 08:22:58.027720] [0x00007f131a7e4700] [info]    [OpenAuto] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-04-25 08:22:58.027829] [0x00007f131a7e4700] [info]    [OpenAuto] [SensorService] stop.
[2018-04-25 08:22:58.027912] [0x00007f131a7e4700] [info]    [OpenAuto] [VideoService] stop.
[2018-04-25 08:22:58.028063] [0x00007f131a7e4700] [info]    [OpenAuto] [BluetoothService] stop.
[2018-04-25 08:22:58.028166] [0x00007f131a7e4700] [info]    [OpenAuto] [InputService] stop.
[2018-04-25 08:22:58.028247] [0x00007f131a7e4700] [info]    [OpenAuto] [InputDevice] stop.
[2018-04-25 08:22:58.028954] [0x00007f131afe5700] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 30, native code: 0
[2018-04-25 08:22:58.029145] [0x00007f131afe5700] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2018-04-25 08:22:58.029242] [0x00007f131afe5700] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2018-04-25 08:22:58.029339] [0x00007f131afe5700] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SYSTEM_AUDIO
[2018-04-25 08:22:58.029439] [0x00007f131a7e4700] [info]    [OpenAuto] [App] Waiting for device...
[2018-04-25 08:22:58.029570] [0x00007f131bfe7700] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 30, native code: 0
[2018-04-25 08:22:58.029747] [0x00007f131a7e4700] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.
[2018-04-25 08:22:58.030021] [0x00007f131afe5700] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 30, native code: 0

f1xpl, 2018-04-25:
Please see this issue https://github.com/f1xpl/openauto/issues/39

jackongithub, 2018-04-25:
Thanks
I ran **sudo apt remove mesa-va-drivers** but that didn't fix it. After installing
**sudo apt-get install gstreamer1.0-plugins-bad** everything works now 😃
Just for information this is running on an intel nuc dn2820fykh.
Thanks
-Jack

closed 2018-04-25 by jackongithub