f1xpl/openauto #82: During openauto compiling receive error

rmiles7721, 2018-05-04:
[  2%] Automatic moc, uic and rcc for target btservice
[  2%] Built target btservice_automoc
[ 14%] Built target btservice
[ 17%] Automatic moc, uic and rcc for target autoapp
[ 17%] Built target autoapp_automoc
[ 19%] Building CXX object CMakeFiles/autoapp.dir/src/autoapp/App.cpp.o
/home/pi/openauto/src/autoapp/App.cpp:20:40: fatal error: f1x/aasdk/USB/AOAPDevice.hpp: No such file or directory
 #include <f1x/aasdk/USB/AOAPDevice.hpp>
                                        ^
compilation terminated.
CMakeFiles/autoapp.dir/build.make:62: recipe for target 'CMakeFiles/autoapp.dir/src/autoapp/App.cpp.o' failed
make[2]: *** [CMakeFiles/autoapp.dir/src/autoapp/App.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/autoapp.dir/all' failed
make[1]: *** [CMakeFiles/autoapp.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

f1xpl, 2018-05-04:
You did not build aasdk.

closed 2018-05-04 by f1xpl

rmiles7721, 2018-05-04:
Figured it out. Thought I had build aasdk properly until I took another look. I copied and pasted part where you build the make files.

cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="/home/pi/aasdk/include" -DAASDK_LIBRARIES="/home/pi/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="/home/pi/aasdk_build" -DAASDK_PROTO_LIBRARIES="/home/pi/aasdk/lib/libaasdk_proto.so" ../openauto

There are spaces between the dashes that shouldn't be there do to the html I guess. That's what I get for not typing it in myself.

tman1982, 2019-03-04:
> Figured it out. Thought I had build aasdk properly until I took another look. I copied and pasted part where you build the make files.
>
> cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE [...]

I'm getting the same original error. I checked to ensure there were no spaces after the '-' dashes... aasdk appears to have built fine. That file it is referencing is not in the aasdk folders though, the path is but the referenced AOAPDevice.hpp is not. Anything else to try? Should I delete the build folder and try building aasdk again?