f1xpl/openauto #84: Black and white video output

jackongithub, 2018-05-09:
Hi
I had openauto working but had to reinstall ubuntu 18.04 LTS.
I re-did everything from scratch following the instructions.
Everything is working fine (phone connects etc) but the screen is black and white with no color at all.
I have attached a screenshot and logs. I have checked and made sure to install everything related to gstreamer because that looks like the issue. But I wanted to get your opinion on it.
[image]

**uname -a**
Linux mini-desktop 4.15.0-20-generic #21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

**console output**
[2018-05-09 21:18:11.587986] [0x00007fecb002f840] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-05-09 21:18:11.626252] [0x00007fec9fe57700] [info]    [OpenAuto] [App] Waiting for device...
[2018-05-09 21:18:12.630675] [0x00007fec9e654700] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-05-09 21:18:21.810761] [0x00007fec9e654700] [info]    [OpenAuto] [App] Device connected.
[2018-05-09 21:18:21.817809] [0x00007fecb002f840] [debug]   [OpenAuto] [AudioInput] create.
[2018-05-09 21:18:21.829649] [0x00007fecb002f840] [debug]   [OpenAuto] [QtVideoOutput] create.
[2018-05-09 21:18:22.190675] [0x00007fec9e654700] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-05-09 21:18:22.190833] [0x00007fec9e654700] [info]    [OpenAuto] [AudioInputService] start.
[2018-05-09 21:18:22.191063] [0x00007fec9e654700] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-05-09 21:18:22.191144] [0x00007fec9e654700] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-05-09 21:18:22.191218] [0x00007fec9e654700] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-05-09 21:18:22.191302] [0x00007fec9e654700] [info]    [OpenAuto] [SensorService] start.
[2018-05-09 21:18:22.191375] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] start.
[2018-05-09 21:18:22.191448] [0x00007fec9e654700] [info]    [OpenAuto] [BluetoothService] start.
[2018-05-09 21:18:22.191523] [0x00007fec9e654700] [info]    [OpenAuto] [InputService] start.
[2018-05-09 21:18:22.197633] [0x00007fec9e654700] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.4, status: 0
[2018-05-09 21:18:22.197677] [0x00007fec9e654700] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-05-09 21:18:22.208768] [0x00007fec9fe57700] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2018-05-09 21:18:22.220471] [0x00007fec9fe57700] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-05-09 21:18:22.233528] [0x00007fec9ee55700] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-05-09 21:18:22.233857] [0x00007fec9ee55700] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-05-09 21:18:22.250226] [0x00007fec9e654700] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: Google Pixel 2
[2018-05-09 21:18:22.250357] [0x00007fec9e654700] [info]    [OpenAuto] [AudioInputService] fill features.
[2018-05-09 21:18:22.250439] [0x00007fec9e654700] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-05-09 21:18:22.250516] [0x00007fec9e654700] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-05-09 21:18:22.250590] [0x00007fec9e654700] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-05-09 21:18:22.250662] [0x00007fec9e654700] [info]    [OpenAuto] [SensorService] fill features.
[2018-05-09 21:18:22.250735] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] fill features.
[2018-05-09 21:18:22.250809] [0x00007fec9e654700] [info]    [OpenAuto] [BluetoothService] fill features
[2018-05-09 21:18:22.250877] [0x00007fec9e654700] [info]    [OpenAuto] [InputService] fill features.
[2018-05-09 21:18:22.321807] [0x00007fec9e654700] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2018-05-09 21:18:22.321854] [0x00007fec9e654700] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2018-05-09 21:18:22.328953] [0x00007fec9ee55700] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2018-05-09 21:18:22.328999] [0x00007fec9ee55700] [info]    [OpenAuto] [AudioInputService] open status: 0
[2018-05-09 21:18:22.329343] [0x00007fec9f656700] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-05-09 21:18:22.329401] [0x00007fec9f656700] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel count: 2
[2018-05-09 21:18:22.330277] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2018-05-09 21:18:22.330357] [0x00007fec9fe57700] [debug]   [OpenAuto] [AudioService] channel: SPEECH_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-05-09 21:18:22.330853] [0x00007fec9ee55700] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2018-05-09 21:18:22.330905] [0x00007fec9ee55700] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-05-09 21:18:22.331924] [0x00007fec9e654700] [info]    [OpenAuto] [SensorService] open request, priority: 0
[2018-05-09 21:18:22.332009] [0x00007fec9e654700] [info]    [OpenAuto] [SensorService] open status: 0
[2018-05-09 21:18:22.335187] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] open request, priority: 0
[2018-05-09 21:18:22.335248] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] open status: 0
[2018-05-09 21:18:22.335850] [0x00007fec9e654700] [info]    [OpenAuto] [InputService] open request, priority: 0
[2018-05-09 21:18:22.336824] [0x00007fec9e654700] [info]    [OpenAuto] [InputService] open status: 0
[2018-05-09 21:18:22.337507] [0x00007fec9e654700] [info]    [OpenAuto] [SensorService] sensor start request, type: 13
[2018-05-09 21:18:22.337948] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] setup request, config index: 3
[2018-05-09 21:18:22.338610] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] setup status: 2
[2018-05-09 21:18:22.344394] [0x00007fec9e654700] [info]    [OpenAuto] [InputService] binding request, scan codes count: 0
[2018-05-09 21:18:22.344632] [0x00007fec9e654700] [info]    [OpenAuto] [InputDevice] start.
[2018-05-09 21:18:22.344758] [0x00007fec9e654700] [info]    [OpenAuto] [InputService] binding request, status: 0
[2018-05-09 21:18:22.344971] [0x00007fec9e654700] [info]    [OpenAuto] [VideoService] video focus indication.
[2018-05-09 21:18:22.346147] [0x00007fec9e654700] [info]    [OpenAuto] [SensorService] sensor start request, type: 10
[2018-05-09 21:18:22.359378] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] open status: 0, channel: SPEECH_AUDIO
[2018-05-09 21:18:22.360184] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] setup request, channel: SPEECH_AUDIO, config index: 1
[2018-05-09 21:18:22.360229] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] setup status: 2, channel: SPEECH_AUDIO
[2018-05-09 21:18:22.362319] [0x00007fec9f656700] [info]    [OpenAuto] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2018-05-09 21:18:22.363743] [0x00007fec9ee55700] [info]    [OpenAuto] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[... 6 lines cut ...]
(autoapp:1486): GLib-GObject-WARNING **: 21:18:23.952: value "-0.000000" of type 'gfloat' is invalid or out of range for property 'contrast' of type 'gfloat'
[2018-05-09 21:19:28.112962] [0x00007fec9ee55700] [info]    [OpenAuto] [AndroidAutoEntity] Shutdown request, reason: 1
[2018-05-09 21:19:28.113918] [0x00007fec9fe57700] [info]    [OpenAuto] [App] quit.
[2018-05-09 21:19:28.114030] [0x00007fec9fe57700] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2018-05-09 21:19:28.114137] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioInputService] stop.
[2018-05-09 21:19:28.114300] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] stop, channel: MEDIA_AUDIO
[2018-05-09 21:19:28.119980] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] stop, channel: SPEECH_AUDIO
[2018-05-09 21:19:28.123388] [0x00007fec9fe57700] [info]    [OpenAuto] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-05-09 21:19:28.127279] [0x00007fec9fe57700] [info]    [OpenAuto] [SensorService] stop.
[... cut ...]
[2018-05-09 21:19:28.130064] [0x00007fec9e654700] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SYSTEM_AUDIO
[2018-05-09 21:19:28.130130] [0x00007fec9fe57700] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.
[2018-05-09 21:19:28.131257] [0x00007fec9fe57700] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 30, native code: 0

** (autoapp:1486): CRITICAL **: 21:19:28.174: gst_vaapi_window_x11_new_with_xid: assertion 'xid != None' failed

parkerlreed, 2018-05-09:
Uninstall gstreamer vaapi

jackongithub, 2018-05-10:
Yep that was it.
Thank you so much :)
-Jack

closed 2018-05-10 by jackongithub