f1xpl/openauto #85: Phone will not connect to Open Auto with cable, works on wireless mode.

ctroxtell, 2018-05-13:
Anyone have any issues getting it to connect to there Raspberry Pi? I've tried 4 different usb cables each time I plug in the cable to the phone the phone enters charging mode then disconnects and keeps doing this about every second. Wireless mode works but lags at times making the music skip, so I also attempted to get Crankshaft to work but Crankshaft does the same thing with the usb cables. Using a Anker 5v 2.4a car charger to power the RP3B+. Is it possible the wire mode is pulling more power and causing this issue?

KLVN, 2018-05-13:
I have the same issue with my RP1B (sometimes lagging, but it works btw). Wireless connection works perfect, but via cable nothing happens.

mbreevoort, 2018-05-13:
Take a look at the tips in this issue #76 https://github.com/f1xpl/openauto/issues/76
try adding this to the config.txt file : max_usb_current=1
gpu_mem=256

ctroxtell, 2018-05-13:
Attempted adding that line to the config.txt file, still continues to do the same thing. About every second I get the "Removable media is inserted" notification in OpenAuto it'll pop up then disappear every second. If I close OpenAuto the "Removable media is inserted" window will pop up and ask what action to so with it. So it appears Open Auto doesn't like something.

RcloP97, 2018-05-13:
Have you all allowed usb permissions? https://github.com/f1xpl/openauto/wiki/udev-rules-(USB-permissions)

ctroxtell, 2018-05-13:
I ran 
sudo wget -P /etc/udev/rules.d https://raw.githubusercontent.com/snowdream/51-android/master/51-android.rules should that of allowed the USB or did I misread? Thanks

ctroxtell, 2018-05-13:
Just ran those commands, restarted...made no difference.

RcloP97, 2018-05-13:
@ctroxtell are you running openauto with a push button?

ctroxtell, 2018-05-14:
No, running OpenAuto via autorun.

RcloP97, 2018-05-14:
Maybe the problem is not from the raspy; try to change the usb roll in the phone; i think that's predetermined with charging, try changeing it to transfer files or something like that; if this doesn't work i don't know what could be the problem

ctroxtell, 2018-05-14:
It is set as MTP but will not work. May try an old Samsung S6 I have later today to see if the issue is with the phone or something else.

RcloP97, 2018-05-14:
Okey; good luck

ctroxtell, 2018-05-15:
Okay, tried the Galaxy S6 with no luck. The S6 has a notification it connected to Android Auto when the USB cable is plugged in. But then nothing happens on the Pi except the Pi will not accept touches until the S6 is disconnected. So it does a small bit more I guess, but still no luck. Is there a log file kept on the Pi?

RcloP97, 2018-05-15:
I have the same issue when i try to run openauto by the push button; is it right that you can open the main GUI and be able to change the settings..etc until the phone is connected? And then ita connected nothing happens, it keeps the same screen but the touch is not recognised? Days ago i had this issue; i fibally assumed that the failure was from the script i made for my push buttons (should work same with autorun; maybe the language of the script changes) so, nowadays i have my project a little bit pulled apart (final exams from University are approaching) but if you whant to run your openauto without the autorun,  should work, by typing (if you have followed the build instructions)
 /home/pi/openauto/bin/autoapp
If this doesn't work reinstall openauto (I personally recommend formatting your sd card, installing your S.O. and then install again the openauto)

ctroxtell, 2018-05-16:
Yeah, that's pretty much what is happening. Opening via file explorer yields the same results. Haven't tried running it via terminal, not sure if that will yield a different result.

mbreevoort, 2018-05-16:
@ctroxtell to troubleshoot this try to run in a terminal (where you installed openauto)
/home/pi/openauto/bin/autoapp > openauto.log
* Connect the phone
* exit openauto 
* check the openauto.log for errors (cat openauto.log)
Step 2: If it is a permissions problem (or not) try to run it as root
sudo /home/pi/openauto/bin/autoapp > openauto.log
If it works then you have a permissions problem....

ctroxtell, 2018-05-17:
I got the libEGL warning plus a message stating "** (Auto app:935): Warning**: Error retrieveing accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ally.Bus was not provided by any .service files

ctroxtell, 2018-05-19:
Okay, finally just got a solution to this problem. Hope it helps someone else out down the road. Had to go to Settings>Personalization>Miscellaneous>MTP instead charging. After disabling this option Android Auto connects right up.

closed 2018-05-19 by ctroxtell