f1xpl/openauto #86: Charging question

dj21d, 2018-05-16:
Not so much an issue but a question about phone charging while connected to your raspberry pi. Is connecting through USB adequate enough for charging a newer phone? If not what is the recommendation for being able to charge your phone while connected? I've read up on wireless mode but my car does not have wifi.

St3v390, 2018-05-16:
Hey,
It is depending on how much Amper your device needs to charge. If you set "max_usb_current=1" in "/boot/config.txt" the USB will have 1.2A.
For the wireless mode it's not necessary that your car has wifi. I'm not sure so please correct me if I am wrong, but as I know you have to start a wifi Hotspot on your fone and connect the PI to it.

dj21d, 2018-05-16:
Thanks for the reply @St3v390.
I find with my s6 that charging with less than 2A only maintains charge but doesn't actually increase the battery. 

The hot spot mode is neat, I could probably automate my phone to turn it off/on when I enter my car, but I do find my phone really heats up with the hotspot activated. It's a bit concerning. 
These are all personal problems I know, just interested to see what the rest of the community is doing.

dj21d, 2018-05-16:
@KLVN Didn't know that was a thing, thank you!
If this is something you use how long does it usually take from power on for the openauto/wifi to become available?

KLVN, 2018-05-23:
@dj21d Haven't tried it yet with an access point, but it's likely that I will go this way. My Raspberry Pi 1 is pretty slow, so booting and launching OpenAuto takes up to ~2 minutes.

cezarylaksa, 2018-06-03:
@dj21d 5V 2A and not charging? Have you measured this with multimeter or something like keweisi? For me it sounds like bad/too long usb cable and/or bad charger. 
About charging phone while using usb connection to RPi - you can try powered usb data + charging hub. I'm using regular usb 3.0 powered hub and it charges my phone while not taking power directly from RPi.

closed 2018-07-10 by f1xpl