f1xpl/openauto #89: USB Mode

AlaricAU, 2018-05-21:
I've not been able to nail down whats causing it, but when I connect my phone, android auto launches, but openauto doesn't leave the waiting menu.

Very briefly it will connect in data mode, but then quickly switches to "charge phone only". If I change this option it just reverts back to the charging option.

Is there some sort of debug somewhere I can check?

AlaricAU, 2018-05-21:
Further to the above, I plugged in a phone that did not have android auto installed. It connected fine as a MTP device. I then put Android Auto on it, and it is now having the same problem. After uninstalling Android Auto, the issue remains, and google play services displays an Android Auto notification.

I think two things are happening:
1)Android Auto is causing the phone to switch to an unlisted connection mode or something. This may be functioning correctly.
2)OpenAuto is not detecting the phone correctly.

f1xpl, 2018-05-22:
Please check for power issues. It looks like RPI USB disconnects your phone due to high power consumption.

Also according to [this](https://github.com/f1xpl/openauto/issues/85) issue, for some phones MTP must be enabled in order to use Android Auto.