f1xpl/openauto #90: Quit OpenAuto app not available

Sonnet71, 2018-05-29:
Thanks for the building instructions and the given USB rules. All went well and OpenAuto is up and running. But I found no way to quit this app to use all the other functionalities of the basic Raspian system. It just stays in fullscreen mode and the PI system is not accessible.

When pressing the button in the low right corner on real AndroidAuto headunits I'm getting the option to quit an go back to the nativ screen of the radio. In OpenAuto I just get a new screen with a large button named with "Return to OpenAuto"

Any help on this?

f1xpl, 2018-05-30:
https://github.com/f1xpl/openauto/issues/33

f1xpl, 2018-09-11:
Mentioned functionalities are part of OpenAuto Pro (new project that I am involved in). Take a look at the https://www.youtube.com/watch?v=9sTOMI1qTiA to see if this is something you are looking for.

closed 2019-01-17 by f1xpl