f1xpl/openauto #91: Errors while building openauto and then could not connect to display

synonymousz, 2018-05-31:
These are the errors I get while building openauto:

In file included from /home/pi/aasdk/include/f1x/aasdk/USB/AOAPDevice.hpp:23:0,
                 from /home/pi/openauto/src/autoapp/App.cpp:20:
/usr/include/libusb-1.0/libusb.h:736:4: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:767:4: warning: ISO C++ forbids zero-size array ‘dev_capability’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:1263:4: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
___________________________________________________________________________________

In file included from /opt/vc/include/interface/vcos/vcos.h:144:0,
                 from /opt/vc/include/interface/vmcs_host/vc_dispmanx.h:33,
                 from /opt/vc/include/bcm_host.h:50,
                 from /home/pi/openauto/src/autoapp/Projection/OMXVideoOutput.cpp:23:
/opt/vc/include/interface/vcos/vcos_logging.h:225:39: warning: ISO C++ does not permit named variadic macros [-Wvariadic-macros]
 #  define _VCOS_LOG_X(cat, _level, fmt...)   do { if (vcos_is_log_enabled(cat,_level)) vcos_log_impl(cat,_level,fmt); } while (0)
                                       ^~~
/home/pi/openauto/src/autoapp/Projection/OMXVideoOutput.cpp: In member function ‘bool f1x::openauto::autoapp::projection::OMXVideoOutput::createComponents()’:
/home/pi/openauto/src/autoapp/Projection/OMXVideoOutput.cpp:207:194: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wpedantic]
 ", static_cast<ILCLIENT_CREATE_FLAGS_T>(ILCLIENT_DISABLE_ALL_PORTS | ILCLIENT_ENABLE_INPUT_BUFFERS)) != 0)
                                                                                                    ^
/home/pi/openauto/src/autoapp/Projection/OMXVideoOutput.cpp:213:125: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wpedantic]
 mponent(client_, &components_[VideoComponent::RENDERER], "video_render", ILCLIENT_DISABLE_ALL_PORTS) != 0)
                                                                                                    ^
/home/pi/openauto/src/autoapp/Projection/OMXVideoOutput.cpp:219:115: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wpedantic]
 _create_component(client_, &components_[VideoComponent::CLOCK], "clock", ILCLIENT_DISABLE_ALL_PORTS) != 0)
                                                                                                    ^
/home/pi/openauto/src/autoapp/Projection/OMXVideoOutput.cpp:231:129: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wpedantic]
 ent(client_, &components_[VideoComponent::SCHEDULER], "video_scheduler", ILCLIENT_DISABLE_ALL_PORTS) != 0)
         
____________________________________________________________________________________

In file included from /home/pi/aasdk/include/f1x/aasdk/USB/AOAPDevice.hpp:23:0,
                 from /home/pi/openauto/src/autoapp/Service/AndroidAutoEntityFactory.cpp:19:
/usr/include/libusb-1.0/libusb.h:736:4: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:767:4: warning: ISO C++ forbids zero-size array ‘dev_capability’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:1263:4: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^

________________________________________________________________________________________

In file included from /opt/vc/include/interface/vcos/vcos.h:144:0,
                 from /opt/vc/src/hello_pi/libs/ilclient/ilclient.h:47,
                 from /home/pi/openauto/include/f1x/openauto/autoapp/Projection/OMXVideoOutput.hpp:24,
                 from /home/pi/openauto/src/autoapp/Service/ServiceFactory.cpp:34:
/opt/vc/include/interface/vcos/vcos_logging.h:225:39: warning: ISO C++ does not permit named variadic macros [-Wvariadic-macros]
 #  define _VCOS_LOG_X(cat, _level, fmt...)   do { if (vcos_is_log_enabled(cat,_level)) vcos_log_impl(cat,_level,fmt); } while (0)
                                       ^~~

___________________________________________________________________________________________

In file included from /home/pi/aasdk/include/f1x/aasdk/USB/IUSBWrapper.hpp:24:0,
                 from /home/pi/aasdk/include/f1x/aasdk/USB/IUSBHub.hpp:23,
                 from /home/pi/aasdk/include/f1x/aasdk/USB/USBHub.hpp:23,
                 from /home/pi/openauto/src/autoapp/autoapp.cpp:21:
/usr/include/libusb-1.0/libusb.h:736:4: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:767:4: warning: ISO C++ forbids zero-size array ‘dev_capability’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^
/usr/include/libusb-1.0/libusb.h:1263:4: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]
  [0] /* non-standard, but usually working code */
    ^

____________________________________________________________________________________________

Everything else is fine, just copied out the errors.
When I then try to start openauto this error appears:
QXcbConnection: Could not connect to display 
Aborted

I was using a Raspberry Pi 3 B+ with Raspbian Stretch on 32gb SD-Card. I did nothing else from here than trying to do the openauto stuff. (just activated ssh, connected and tried the build instructions)

I hope you can help me to use this awesome project. Looking forward for your answers.
Thanks

f1xpl, 2018-05-31:
Are you using Raspbian Stretch Desktop or Lite? Did you try to run OpenAuto directly on RPI (not via ssh)?

synonymousz, 2018-05-31:
Desktop
I did not try to run it without ssh yet

synonymousz, 2018-05-31:
I just tried with keyboard and it started. Can this issue leave open for me if i have more problems or shall i open new issues?

synonymousz, 2018-05-31:
Now I am experiencing the problem that I am connected to the Android Auto, but when i say okay Google,it only shows Up but doesnt do the command i say afterwards

Bobet21, 2018-09-22:
what was the cause of the error finally? I am running strech lite, and I am facing the same could not connect to display error

synonymousz, 2018-09-22:
I got the error when trying to start over SSH. When I connected my keyboard to the Pi, it gave some warnings but it worked

Bobet21, 2018-09-22:
mmm ok, i actually see this while beeing connected with my keayboard to the Pi...

Bobet21, 2018-09-22:
https://github.com/wkhtmltopdf/wkhtmltopdf/issues/2037#issuecomment-62019521 is it related to the same package ? seems that there is some configuration mismtacth ? i'll open a new issue for this...