f1xpl/openauto #95: OpenAuto in Window Mode / Side-by-side with another application

danielsteiner, 2018-06-05:
Hello there, 

after playing around for some hours tonight, I am still wondering if I can have two applications running on two virtual desktops - so far everything is setup on my RPi3 and I can run OpenAuto and the other application without any problem but I need to run both at the same time and be able to switch between them. Any idea how to do that? I am currently thinking about a button that would switch the active workspace but . Alternatively, a window mode would be working as well. 
Anyone got an idea how to do that?

RcloP97, 2018-06-06:
Hi there Daniel; im doing something pretty similar to you; i think no; is not possible to switch between openauto and another application running; you should kill the openauto process in order to go to the other one app; and yes i have done it with two buttons; i still have the algorithm in development; please answer something in order to remind me this Friday to send you the little program

St3v390, 2018-06-06:
Hi Daniel,
I was trying the same. So I asked the same question on Gitter and received an answered from the developer.
He told me that this doesn't work but he will try to implement this function in the future.

danielsteiner, 2018-06-06:
Hmm killing OpenAuto and restarting it is not an option for me as this kills the Audioplayback via Bluetooth.  - I will take a look on how viktorgino is doing that in headunit-desktop as his solution allowed you to switch between DAB/AC/OpenAuto.

vapor99, 2018-06-20:
Just built an infotainment system in my car with Raspbian running OpenAuto. I would love to have my desktop apps open next to openauto.  :)

Bobet21, 2018-09-21:
Did any one of you tried to have 2 monitor connected ?

HazWard, 2018-11-28:
To run the application in windowed mode, you need to do these changes to the source code and rebuild
```patch
diff --git a/src/autoapp/Projection/QtVideoOutput.cpp b/src/autoapp/Projection/QtVideoOutput.cpp
index ad4636c..9aa9107 100644
--- a/src/autoapp/Projection/QtVideoOutput.cpp
+++ b/src/autoapp/Projection/QtVideoOutput.cpp
@@ -73,7 +73,7 @@ void QtVideoOutput::onStartPlayback()
     videoWidget_->setAspectRatioMode(Qt::IgnoreAspectRatio);
     videoWidget_->setFocus();
     videoWidget_->setWindowFlags(Qt::WindowStaysOnTopHint);
-    videoWidget_->setFullScreen(true);
+    videoWidget_->setFullScreen(false);
     videoWidget_->show();
 
     mediaPlayer_->setVideoOutput(videoWidget_.get());
diff --git a/src/autoapp/autoapp.cpp b/src/autoapp/autoapp.cpp
index 2aa9dcb..3746bb5 100644
--- a/src/autoapp/autoapp.cpp
+++ b/src/autoapp/autoapp.cpp
@@ -99,7 +99,7 @@ int main(int argc, char* argv[])
     connectDialog.setWindowFlags(Qt::WindowStaysOnTopHint);
 
     QObject::connect(&mainWindow, &autoapp::ui::MainWindow::exit, []() { std::exit(0); });
-    QObject::connect(&mainWindow, &autoapp::ui::MainWindow::openSettings, &settingsWindow, &autoapp::ui::SettingsWindow::showFullScreen);
+    QObject::connect(&mainWindow, &autoapp::ui::MainWindow::openSettings, &settingsWindow, &autoapp::ui::SettingsWindow::showNormal);
     QObject::connect(&mainWindow, &autoapp::ui::MainWindow::openConnectDialog, &connectDialog, &autoapp::ui::ConnectDialog::exec);
 
     qApplication.setOverrideCursor(Qt::BlankCursor);
@@ -108,7 +108,7 @@ int main(int argc, char* argv[])
         qApplication.setOverrideCursor(cursor);
     });
 
-    mainWindow.showFullScreen();
+    mainWindow.showNormal();
 
     aasdk::usb::USBWrapper usbWrapper(usbContext);
     aasdk::usb::AccessoryModeQueryFactory queryFactory(usbWrapper, ioService);

```

Bobet21, 2018-11-28:
thanks @HazWard What happen to android auto in that configuration? Does it go full screen anyhow?

HazWard, 2018-11-28:
If all these changes are applied, every window opens in windowed mode.

Later on, someone could use something like a window manager to have workspaces for different applications and used events (button pressed, new window appearing, etc) to switch between workspaces. The only thing missing IMO is being able to configure what happens when the "Return to OpenAuto" button is pressed inside Android Auto (it could possibly redirect to actions on the system).

Bobet21, 2018-11-28:
Which file did you modify? I am not clear or missing something there...

Bobet21, 2018-11-29:
I have now troubled to rebuidl - hope to be able to do it soon - can we have two version/build of openauto co-exisiting?

Bobet21, 2018-11-30:
It's working fine and work well as window management - now i need to connect to AndroidAuto - i suspect and issue with the initial build i have done before - would need to restart from scratch. @HazWard thanks a lot would you have a screen shot of Androidauto in window mode while I am debugging my phone connection ?

HazWard, 2018-12-03:
[image]

Bobet21, 2018-12-03:
Yep, thanks @HazWard , are you able to plug in Android Auto as well ? In my case it seems to be the one thing causing problem, it still goes in full screen.

Bobet21, 2018-12-03:
Thanks, which OS are you using?

HazWard, 2018-12-08:
Arch Linux

HazWard, 2019-02-08:
It's running on a laptop, on a Raspberry Pi you'll probably have to install the OS and OpenAuto manually (most likely by compiling OpenAuto separately and copying it over to the Pi). This way you can install a tiling window manager and probably have a script to open OpenAuto at startup.

Bobet21, 2019-02-12:
Thanks, I installed archlinux on the raspberry PI, and Buidl openauto without issue, but android auto still opens in full screen inside i3. I don't think this is feasible on the RPI.

HazWard, 2019-02-16:
> What benefit do you expect by building OA separately and copying it over to the PI?

I didn't expect the Pi to be able to compile OA in a timely manner but I guess I was wrong.
Yo can set a rule inside i3 to force OA to start in tiling mode. Unfortunately I don't have the info but the [user guide](https://i3wm.org/docs/userguide.html) is pretty good. If I find any details on how to do this I'll link it here.